openrazer/openrazer #1162: Exceptions Raised on Fresh Installation

ncabete, 2020-06-04:
### Distribution

Ubuntu 18.04 Bionic

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Okay, so after a fresh installation of OpenRazer + Polychromatic, everything was working fine until I decided to change a lighting profile. After that, the daemon crashed and it did not allow me to make any further changes.
Rebooting my computer brings the daemon back up, but once again any change I make shuts it down and prevents me from enabling it back up again (unless I reboot again)

**Output of: openrazer-daemon -Fv:**
```
Starting daemon.
2020-06-04 20:55:28 | razer                          | INFO     | Initialising Daemon (v2.8.0). Pid: 12130
2020-06-04 20:55:28 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2020-06-04 20:55:28 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2020-06-04 20:55:28 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2020-06-04 20:55:28 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus2000 ------------------------ (1532:005E)
2020-06-04 20:55:28 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEliteDVaEdition ------------- (1532:006A)
(removed similar lines as they're not relevant & would make this issue gigantic)
2020-06-04 20:55:28 | razer                          | INFO     | Found device.0: 0003:1532:0226.0004
2020-06-04 20:55:28 | razer.device0                  | INFO     | Initialising device.0 RazerHuntsmanElite
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/daemonize.py", line 245, in start
    self.action(*privileged_action_result)
  File "/usr/bin/openrazer-daemon", line 125, in run_daemon
    test_dir=args.test_dir)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/daemon.py", line 107, in __init__
    self._load_devices(first_run=True)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/daemon.py", line 401, in _load_devices
    razer_device = device_class(sys_path, device_number, self._config, testing=self._test_dir is not None, additional_interfaces=sorted(additional_interfaces))
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/keyboards.py", line 54, in __init__
    super(_RippleKeyboard, self).__init__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/keyboards.py", line 23, in __init__
    super(_MacroKeyboard, self).__init__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 482, in __init__
    super().__init__(device_path, device_number, config, testing, additional_interfaces, additional_methods)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 445, in __init__
    super().__init__(device_path, device_number, config, testing, additional_interfaces, additional_methods)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 88, in __init__
    DBusService.__init__(self, self.BUS_PATH, object_path)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/service.py", line 87, in __init__
    super(DBusService, self).__init__(bus_object, object_path)
  File "/usr/lib/python3/dist-packages/dbus/service.py", line 455, in __init__
    validate_object_path(object_path)
ValueError: Invalid object path '/org/razer/device/J': contains invalid character ''

Stopping daemon.
Exception ignored in: <bound method RazerDevice.__del__ of RazerHuntsmanElite:J>
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 431, in __del__
    self.close()
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 345, in close
    self._close()
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/keyboards.py", line 59, in _close
    super(_RippleKeyboard, self)._close()
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/keyboards.py", line 43, in _close
    self.key_manager.close()
AttributeError: 'RazerHuntsmanElite' object has no attribute 'key_manager'
```

chabad360, 2020-06-07:
What is your system's language (`contains invalid character '�'`)? You can check the `LANG` and fam variables, for the answer.

ncabete, 2020-06-08:
> What is your system's language (`contains invalid character '�'`)? You can check the `LANG` and fam variables, for the answer.

Portuguese (pt-PT). Is there a specific font I need to install?

z3ntu, 2020-06-12:
@ncabete Please try running `hexdump -C /sys/bus/hid/drivers/razerkbd/0*/device_serial` and sending the result.

closed 2020-07-18 by ncabete