openrazer/openrazer #1166: Naga trinity is not fully working

rkmax, 2020-06-06:
### Distribution

- ArchLinux kernel 5.6.15
- Openrazer 2.8.0-1
- Razer naga trinity (1532:0067)

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

## Buttons dont work properly

Currently I'm using the seven buttons face, some buttons like 1,3,4,6,7 are working (is like pressing the numbers of the keyboard I think is fine) except 2 and 5 button, xev show nothing when are pressed and configure something in a game also doesnt respond

## lighting, polling, dpi and brightness are not working properly

The values are not shown when I open the app polychromatic controller and lighting effect only static is available

- Any clue how to solve the problems?

Techwolf, 2020-06-08:
I had the same error in dmesg, I ended up pulling the buttons off and placing it back on fixed it. Note I did this while the mouse was plug into the USB port.

z3ntu, 2020-06-12:
> The values are not shown when I open the app polychromatic controller 

If you mean that you set a value, then re-open polychromatic and the value is displayed differently, that's a limitation with openrazer which should be solved soon^tm

> and lighting effect static it's the only one available

This is the only effect which was added as a part of #914 . If you find that other effects work (you'll have to edit the source code and rebuild) then pull requests are welcome :)

As I don't see anything other not working I'm going to close the issue.

closed 2020-06-12 by z3ntu