openrazer/openrazer #1169: Macro buttons release themselves when another key is pressed

mads256h, 2020-06-11:
### Distribution

Arch Linux

### Support

- [ x ] Have you tried rebooting?
- [ x ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I am using a blackwidow chroma.

I wanted to remap my m4 key to left ctrl as i damaged it, i got the remap to work with udev hwdb but the left ctrl didnt work. I found the cause when i checked the events in evtest. I held the macro button down and pressed another key on my keyboard and i saw that the macro button released itself just before the other key press even though i did not release the macro button on my keyboard:

```
Event: time 1591903481.720013, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7006b
Event: time 1591903481.720013, type 1 (EV_KEY), code 186 (KEY_F16), value 1
Event: time 1591903481.720013, -------------- SYN_REPORT ------------
Event: time 1591903481.804024, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7006b
Event: time 1591903481.804024, type 1 (EV_KEY), code 186 (KEY_F16), value 0
Event: time 1591903481.804024, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70006
Event: time 1591903481.804024, type 1 (EV_KEY), code 46 (KEY_C), value 1
Event: time 1591903481.804024, -------------- SYN_REPORT ------------

```

chabad360, 2020-06-11:
To put it mildly, there are a lot of issues with the macro system... There's a reason why I'm completely scraping it in #1124...

chabad360, 2020-06-11:
FWI, this would actually be resolved by #845.

closed 2020-12-25 by stale[bot]