openrazer/openrazer #1170: razer atheris BT mouse toggle betweens connected and disconnected

flynneva, 2020-06-13:
### Distribution

Ubuntu 20.04 Focal Fossa

### Support

- [ x ] Have you tried rebooting?
- [ x ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

When connecting the razer atheris mouse like normal (holding the buttons for 3 secs, click connect) everything seems fine. Once selected, the connection then begins to toggle between connected & disconnected indefinitely and the mouse is non-functional.

looking at /var/log/syslog...these errors keep repeating while the bluetooth connection is toggling:

```
un 13 14:47:54 xps bluetoothd[692]: Error reading PNP_ID value: Unexpected error code
Jun 13 14:47:54 xps bluetoothd[692]: Error reading PNP_ID value: Request attribute has encountered an unlikely error
Jun 13 14:47:54 xps bluetoothd[692]: Protocol Mode characteristic read failed: Request attribute has encountered an unlikely error
Jun 13 14:47:54 xps bluetoothd[692]: Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Jun 13 14:47:54 xps bluetoothd[692]: message repeated 5 times: [ Read Report Reference descriptor failed: Request attribute has encountered an unlikely error]
Jun 13 14:47:54 xps bluetoothd[692]: Report Map read failed: Request attribute has encountered an unlikely error
Jun 13 14:47:54 xps bluetoothd[692]: HID Information read failed: Request attribute has encountered an unlikely error
```

z3ntu, 2020-06-26:
Does this issue also happen without openrazer installed?

flynneva, 2020-06-26:
yes, same symptoms and logs as above

parkerlreed, 2020-06-26:
So that wouldn't be an openrazer issue...

flynneva, 2020-06-26:
figured I would create an issue here in case anyone else is trying to also use a razor mouse on Ubuntu 20.04. ill close it though since it doesnt apply

closed 2020-06-26 by flynneva

parkerlreed, 2020-06-27:
Yeah I understand the frustration. I don't have the mouse itself, but looking online I can't find many mentions of issues with that mouse. 

Maybe check out linux_questions on reddit.