openrazer/openrazer #1173: Touchpad buttons stopped working on Ubuntu 20.04 after installation

johann-petrak, 2020-06-17:
I had trouble with the keyboard and touchpad randomly not working at the login screen under Ubuntu 18.04.

I upgraded to 20.04 and both the keyboard and touchpad, including the two touchpad "mouse" buttons worked fine.

After this I installed the openrazer packages and after that, the touchpad buttons stopped working entirely (in the login screen as well as after login). The cursor moves correctly from the touchpad, but pressing either of the two touchpad buttons does nothing. 

I purged the openrazer package and removed the repo but the problem now persists.

nightsky30, 2020-06-17:
I'm wondering if this is #877 except you can't unplug touchpads/keyboards with you Blade to reset things like you can with other peripherals. Does this happen every boot, and can you check logs for similar errors?

johann-petrak, 2020-06-21:
The annoying part is that this seems to be an intermittent problem. Before upgrading to 20.04 I sometimes had trouble using the touchpad at the login screen after a fresh boot, sometimes it worked. 

Now, after installing a system upgrade after removing the openrazer packages, things seem to be back to normal again, but it may be just by chance. So I am not even sure there is a connection with openrazer or not at this point. I will try to go through the whole install/uninstall cycle for openrazer once more and report back.

Where should I look for error messages about the razer devices? Command `dmesg`?

nightsky30, 2020-06-21:
In #877 looks like the errors were presenting themselves in journalctl. Not saying it is definitely that issue, but it could be. When it happens you might check there:

```sudo journalctl -b | grep -i razer_mount```

I've not gotten the errors recently, but also haven't been building the modules as often as I was. When I had the errors they were intermittent as well.

oscarortegano, 2020-08-04:
Same problem happen with external mouse, right click suddenly stop working. Workaround is switching to a virtual terminal with ctrl+alt+fn and go back to desktop environment.

Ubuntu 20.04

johann-petrak, 2020-12-26:
Maybe this was a hardware problem after all, after a while I noticed that the touchpad was bulging out of the notebook and it turned out that the battery has expanded so much that it partly damaged the device. 
Of course Razer does not sell a replacement battery any more. I have ticked off using a Razer device as a mistake and am using a different brand device now.

closed 2021-07-22 by github-actions[bot]