openrazer/openrazer #1174: FN keys doesn't work when the daemon is running

PanosTrak, 2020-06-17:
### Distribution
```
$ uname -r
5.6.16-1-MANJARO
```
```
$ openrazer-daemon --version
openrazer-daemon 2.8.0
```

Keyboard: Huntsman TE

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Everything works fine but the problem is when the ``openrazer-daemon`` is running the FN + [F1-12] doesn't work, its like I am not holding the FN key at all, but when i stop the daemon the keys work fine!

psidhu, 2020-06-26:
Confirmed on my system as well.

```
$ uname -a
Linux devel 5.6.18-200.fc31.x86_64 #1 SMP Wed Jun 10 22:18:55 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

```
$ openrazer-daemon --version
openrazer-daemon 2.8.0
```

Keyboard: Huntsman TE

psidhu, 2020-06-29:
Just a FYI, but I "fixed" it for the huntsman TE only. It's a hardfix, but good enough for me. Hopefully it helps someone else.

```
diff --git a/driver/razerkbd_driver.c b/driver/razerkbd_driver.c
index 909a797..dfa50fb 100644
--- a/driver/razerkbd_driver.c
+++ b/driver/razerkbd_driver.c
@@ -2075,7 +2075,7 @@ static int razer_raw_event(struct hid_device *hdev, struct hid_report *report, u
     }
 
     // The event were looking for is 16 bytes long and starts with 0x04
-    if(intf->cur_altsetting->desc.bInterfaceProtocol == USB_INTERFACE_PROTOCOL_KEYBOARD && size == 16 && data[0] == 0x04) {
+    if(intf->cur_altsetting->desc.bInterfaceProtocol == USB_INTERFACE_PROTOCOL_KEYBOARD && size == 22 && data[0] == 0x04) {
         // Convert 04... to 0100...
         int index = size-1; // This way we start at 2nd last value, does subtract 1 from the 15key rollover though (not an issue cmon)
         u8 cur_value = 0x00;
```

z3ntu, 2020-07-07:
See also #872 
We had a fix in #1049 but unfortunately it introduced a regression so we had to revert the fix: https://github.com/openrazer/openrazer/pull/1049#issuecomment-621868240

PanosTrak, 2020-07-08:
@z3ntu what do you suggest to do for a quick fix if there is not a way for this to be fixed?

z3ntu, 2020-07-09:
Quick fix is https://github.com/openrazer/openrazer/issues/1174#issuecomment-650860866 I guess

mvgian, 2020-09-23:
> Just a FYI, but I "fixed" it for the huntsman TE only. It's a hardfix, but good enough for me. The F11 and F12 KB modifiers stop working, but media keys work which is all I cared about. Hopefully it helps someone else.
> 
> ```
> diff --git a/driver/razerkbd_driver.c b/driver/razerkbd_driver.c
> index 909a797..dfa50fb 100644
> --- a/driver/razerkbd_driver.c
> +++ b/driver/razerkbd_driver.c
> @@ -2075,7 +2075,7 @@ static int razer_raw_event(struct hid_device *hdev, struct hid_report *report, u
>      }
>  
>      // The event were looking for is 16 bytes long and starts with 0x04
> -    if(intf->cur_altsetting->desc.bInterfaceProtocol == USB_INTERFACE_PROTOCOL_KEYBOARD && size == 16 && data[0] == 0x04) {
> +    if(intf->cur_altsetting->desc.bInterfaceProtocol == USB_INTERFACE_PROTOCOL_KEYBOARD && size == 22 && data[0] == 0x04) {
>          // Convert 04... to 0100...
>          int index = size-1; // This way we start at 2nd last value, does subtract 1 from the 15key rollover though (not an issue cmon)
>          u8 cur_value = 0x00;
> ```

Hi. May I know how you apply this fix?

inputsh, 2020-11-17:
@mvgian Follow these instructions: https://github.com/openrazer/openrazer/wiki/Building-a-package

Replace the line starting with the minus with the line starting with the plus before running `./scripts/build_debs.sh`.

psidhu, 2020-12-04:
@z3ntu I'm fairly certain that the proper way for us to fix this is to check size == 16 or size == 22.

What I bet happened is someone at Razer had a fixed size of 16 on fn modifiers in the firmware, but changed that fixed size to 0x16 (22). When I get a sec, I'll submit a PR unless someone else can get to it first.

closed 2021-01-31 by z3ntu