openrazer/openrazer #1180: Razer Cynosa Chroma not recognized

ziovec, 2020-06-24:
### Distribution

Manjaro 20.0.3
Kernel 5.6..16.1

### Support

- [x] Have you tried rebooting?
- [x]  Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I'm using `openrazer-daemon 2.8.0` with polychromatic.
While my keyboard (Cynosa Chroma) is being correctly detected via `lsusb` I can't see it in Polychromatic nor other gui interfaces.
Other devices working fine.
[razer.log](https://github.com/openrazer/openrazer/files/4824980/razer.log)

nightsky30, 2020-06-24:
Also looks like #877

thibaultserti, 2020-07-23:
Exact same problem here with Kernel 5.7.9-1 and Manjaro 20.0.3, device is not recognized.

z3ntu, 2020-07-23:
Yep, closing as duplicate of #877

closed 2020-07-23 by z3ntu