openrazer/openrazer #1181: Razer mamba wireless not detected via the app

YuvalKrispin, 2020-06-25:
I use the mouse as a wired mouse all the time,
And I want to control the colours and the button mapping on the devise,
the mouse is working and the DPI button is working as well,
I have searched in the last couple of weeks but haven't found a fix yet.
I'm new in the Linux Ubuntu community so pleases tall me what I need to add to this Issue

z3ntu, 2020-07-07:
Hi @YuvalKrispin , could you please run the command 
`sudo dkms install openrazer-driver/2.8.0` and send me the output?

okeshawarz, 2020-08-10:
I'm having the exact same problem, with the exact same model mouse. I'm running Manjaro. Here's the output of `sudo dkms install openrazer-driver/2.8.0`:

`Creating symlink /var/lib/dkms/openrazer-driver/2.8.0/source ->
                 /usr/src/openrazer-driver-2.8.0

DKMS: add completed.
Error! echo
Your kernel headers for kernel 5.7.14-1-MANJARO cannot be found at
/usr/lib/modules/5.7.14-1-MANJARO/build or /usr/lib/modules/5.7.14-1-MANJARO/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.`

okeshawarz, 2020-08-10:
Sorry. I should have looked harder at the troubleshooting guide. I needed to install the correct kernel headers. However, in polychromatic, the  mouse is still listed as 'not recognized' in polychromatic.

closed 2021-06-21 by github-actions[bot]