openrazer/openrazer #1182: Razer blade stealth late 2017 keyboard doesnt work

Lillievela, 2020-06-26:
### Distribution

### Support

- [x ] Have you tried rebooting?
- [x ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

z3ntu, 2020-07-07:
Hi @Lillievela , I'm not aware of the driver doing any input "manipulation", especially on the laptops. From what I can read you have already tried to uninstall OpenRazer and then it works again?

Lillievela, 2020-07-07:
Yes, i am able to change the colours and settings, and using my phone with kde connect i can remove openrazer and keep the changes. but its just that i cannot keep it installed as it somehow is preventing my from using the keyboard

z3ntu, 2020-07-08:
Can you try installing an older version (e.g. v2.5.0) and see if that works? You could go as far back as v2.1 which is where support for that device was introduced (see #420, #422)

Lillievela, 2020-07-08:
I managed to get 2.6.0 installed via AUR but it still has the same issue

z3ntu, 2020-07-09:
Then I don't really know what's causing the issues, sorry.

closed 2021-06-21 by github-actions[bot]