openrazer/openrazer #1183: Problem communicating daemon with DeathAdder 3.5G on Ubuntu 20.04 LTS

alexandrepalaoro, 2020-06-27:
Distribution: - Ubuntu 20.04 LTS
Kernel: - 5.4.0-39-generic
Mouse: - DeathAdder 3.5G

- I am using the stable build of OpenRazer and Polychromatic (v2.8.0). 
- I have ran all solutions in the troubleshooting page. 
- I have rebooted, and restarted the daemon. 
- I tried the daily version of both openrazer and polychromatic and it didn't work. 

The Polychromatic (and razergenie) both recognize the mouse and I can toggle the scroll and symbol lights on-and-off. However, i cannot change DPI settings of my mouse. Right now, it says that the DPI is "unknown" and every time I try to mess around with it, it says "there was a problem communicating with the daemon for this device".
Any thoughts?
Thanks in advance!

lah7, 2020-06-27:
Polychromatic doesn't properly support this mouse properly yet. It's due to the fact this device only uses the DPI X value but the program tries to set both X and Y. This is a known bug: https://github.com/polychromatic/polychromatic/issues/209

I'm not sure if that's the same situation for RazerGenie.

You should be able to set the DPI manually using this command:

    python3 -c "import openrazer.client; a = openrazer.client.DeviceManager().devices[0]; a.dpi(1800, None)"

alexandrepalaoro, 2020-06-27:
Given the issue with the DPI values it makes a lot of sense that RazerGenie has a similar issue. It does not show DPI values, and doesn't allow you to change.

I tried the command you sent, but it gave an error:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/openrazer/client/devices/mice.py", line 65, in dpi
    dpi_x, dpi_y = self._dbus_interfaces['dpi'].getDPI()
ValueError: not enough values to unpack (expected 2, got 1)

It gave the same error regardless of the dpi value I chose. Any tips?

Thanks a bunch for your help!

lah7, 2020-06-28:
Ah, my bad. Maybe it's `0` instead of `None` for the Y value:

    python3 -c "import openrazer.client; a = openrazer.client.DeviceManager().devices[0]; a.dpi(1800, 0)"

Or it's `0, 1800` (setting Y instead of X). Bit of guesswork as I don't have any device that has a single DPI value.

This will print the current values:

    python3 -c "import openrazer.client; a = openrazer.client.DeviceManager().devices[0]; print(a.dpi)"

alexandrepalaoro, 2020-06-28:
I've tried setting the X and Y values as you suggested. I've also tried to set both with similar values, and with different values, but the same error keeps popping up.

I tried the to print current values and the same error popped up:

````
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/openrazer/client/devices/mice.py", line 65, in dpi
    dpi_x, dpi_y = self._dbus_interfaces['dpi'].getDPI()
ValueError: not enough values to unpack (expected 2, got 1)
````

I think that this is why the daemon has problems showing AND changing the DPI values: it stops when it gets only one DPI value, regardless of what you are trying to do.

Well, if you (or anyone else) has any other tip I could try, I would appreciate it.
Otherwise, I will close this in a few days.

Thanks for the help!

lah7, 2020-06-28:
Oops, for the setting DPI, I made a syntax error, sorry! :man_facepalming:  For some reason I thought `.dpi` is a function, but it's actually a variable.

    python3 -c "import openrazer.client; a = openrazer.client.DeviceManager().devices[0]; a.dpi = (1800, 0)"

For reading the DPI, perhaps this may work:

    print(a.dpi[0])

The other method is to use a tool called **d-Feet** as the daemon is actually communicating to the driver over D-Bus. This would let you "execute" get/set requests on the **session bus** for **`org.razer`**.

Also, based on this comment: https://github.com/polychromatic/polychromatic/issues/209#issue-471655457, it's the Y value that's `-1`.

alexandrepalaoro, 2020-07-08:
Sorry for the delay in the response.

So, the code doesn't give any error - but it also doesn't change anything. I changed the dpi values to a large range of values, but found no difference

alexandrepalaoro, 2020-07-08:
Sorry for the delay in the response.

So, the code doesn't give any error - but it also doesn't change anything. 
I experimented with different dpi values from a large range of values (e.g., 100 and 3000), but it didn't change anything.
I tried printing the dpi, and it gave the same error (i.e., ValueError: not enough values to unpack (expected 2, got 1)). 

I was almost giving up when I found something that worked for me. I used gnome tweaks to decrease mouse acceleration. With that, I was able to get the mouse sensitivity to a manageable level. So, if anyone has the same problem (and doesn't game seriously), it can be an alternative route.

Thanks for all the help!!

closed 2020-07-08 by alexandrepalaoro