openrazer/openrazer #1184: Support for Razer Kraken Ultimate

32bitflannel, 2020-06-29:
### Device Info

* Name: Razer Kraken Ultimate
* Product Number: ??
* Model Number: RZ04-03180100-R3U1

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
Bus 003 Device 041: ID 1532:0527 Razer USA, Ltd Razer Kraken Ultimate
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0527 
  bcdDevice           50.02
  iManufacturer           1 Razer
  iProduct                2 Razer Kraken Ultimate
  iSerial                 3 00000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0123
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0072
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID            10
        wTerminalType      0x0201 Microphone
        bAssocTerminal          0
        bNrChannels             2
        wChannelConfig     0x0003
          Left Front (L)
          Right Front (R)
        iChannelNames           0 
        iTerminal               6 Sidetone
      AudioControl Interface Descriptor:
[... 254 lines cut ...]
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0025  1x 37 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)

```
Please post the output of
    ls -lah /dev/input/by-id/
```
lrwxrwxrwx 1 root root  10 Jun 25 11:54 usb-Razer_Razer_Kraken_Ultimate_00000000-event-if03 -> ../event13

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

kevchu3, 2020-06-29:
Support for the Kraken Ultimate was added in a recent commit https://github.com/openrazer/openrazer/commit/64eccf4446d8169cd2816a64484136d4e8659a24 although not in the current release v2.8.0.  You can pull in the latest commits by building the openrazer package from source, instructions as follows: https://github.com/openrazer/openrazer/wiki/Building-a-package.  I also have a Kraken Ultimate and this worked for me.  Once they push the next release which includes this commit, you shouldn't have to build from source going forward.

z3ntu, 2020-07-07:
Closing then. Device is already supported.

closed 2020-07-07 by z3ntu