openrazer/openrazer #1185: logo toggling information

ttsesm, 2020-06-30:
Hi guys,

This is not a openrazer per se issue, but I wanted to ask whether it can get in hand.

I have the following issue with my blade https://insider.razer.com/index.php?threads/do-you-guys-get-this-in-your-blades-as-well.55990/ (the razer support doesn't seem to want/care to help solving the issue or maybe they can't). It might be a hardware issue which cannot be solved but it might be possible to address it with some modifications regarding how much current or how often/fast you toggle the logo.

Thus, I was thinking to give a try myself and see whether it would be possible to extract some info regarding the activation/deactivation of the logo and if possible to try to apply any changes myself. Do you think that this would be possible by using the openrazer driver or you do not think it would be possible?

Thanks.

lah7, 2020-07-01:
Unfortunately, it sounds like coil whine. Some of my Razer peripherals make a similar faint high pitch sound when the brightness is on full. OpenRazer doesn't control power parameters like current or voltages, it's just sending commands to the firmware.

I've never seen a Blade laptop in person, but I know OpenRazer does provide an option to turn the logo on/off on some models. Does the logo pulsate, or is it a solid light that's either on or off?

ttsesm, 2020-07-02:
It is coil whine, I got it in both units (original one and its replacement). My guess is due to bad grounding or something. The microphone and logo LED share the same cable going to the display from the motherboard, so it's possible there is a bad grounding of the LED backlight, causing some current to be dumped into your microphones signal wires (causing this high pitch noise). It is showing only when I set the logo toggling to "Breathing" mode from synapse. If I put it to always on or off there is not problem.

In any case thanks for the feedback, if it is not possible to apply controls over the current or the frequency of toggling then I would need to check elsewhere. I guess you can close this issue.

lah7, 2020-07-02:
No worries. If it only happens when it's _breathing_, then it **could** be interference with the USB bus. Razer Synapse actually performs the effects in software, so it essentially "spams" the device with USB commands, whereas OpenRazer sends a command to use the hardware's firmware built-in effect (spectrum, wave, etc) where available.

Though if there's no noise at full brightness and a static colour, then it might not be the same cause of coil whine that I'm experiencing.

closed 2021-06-21 by github-actions[bot]