openrazer/openrazer #1187: Support for Ornata V2

dtx3k, 2020-07-01:
### Device Info

* Name: Ornata V2
* Product Number: RZ03-03380100-R3M1
* Model Number: RZ03-0338

### Device Information
```
Please post the output of
    lsusb -d 1532:025D -v
```

Bus 001 Device 009: ID 1532:025d Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x025d 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Ornata V2
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x006d
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     179
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0016  1x 22 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              2 Razer Ornata V2
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     348
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)

```
Please post the output of
    ls -lah /dev/input/by-id/
```
❯     ls -lah /dev/input/by-id/

total 0
drwxr-xr-x 2 root root 260 jul  1 20:46 .
drwxr-xr-x 4 root root 700 jul  1 20:46 ..
lrwxrwxrwx 1 root root   9 jun 29 19:17 usb-Azurewave_Integrated_Camera_0001-event-if00 -> ../event5
lrwxrwxrwx 1 root root  10 jul  1 20:46 usb-Razer_Razer_Ornata_V2-event-if01 -> ../event23
lrwxrwxrwx 1 root root  10 jul  1 20:46 usb-Razer_Razer_Ornata_V2-event-if03 -> ../event25
lrwxrwxrwx 1 root root  10 jul  1 20:46 usb-Razer_Razer_Ornata_V2-event-kbd -> ../event10
lrwxrwxrwx 1 root root  10 jul  1 20:46 usb-Razer_Razer_Ornata_V2-if01-event-kbd -> ../event11
lrwxrwxrwx 1 root root  10 jul  1 20:46 usb-Razer_Razer_Ornata_V2-if02-event-mouse -> ../event24
lrwxrwxrwx 1 root root   9 jul  1 20:46 usb-Razer_Razer_Ornata_V2-if02-mouse -> ../mouse3
lrwxrwxrwx 1 root root   9 jun 29 19:17 usb-Razer_Razer_Ouroboros-event-mouse -> ../event4
lrwxrwxrwx 1 root root   9 jun 29 19:17 usb-Razer_Razer_Ouroboros-if01-event-kbd -> ../event8
lrwxrwxrwx 1 root root   9 jun 29 19:17 usb-Razer_Razer_Ouroboros-if02-event-kbd -> ../event9
lrwxrwxrwx 1 root root   9 jun 29 19:17 usb-Razer_Razer_Ouroboros-mouse -> ../mouse0

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

not able to capture packets atm

sosik, 2020-08-20:
Hi guys,

I have ornata v2 too. How can I help? Where can I find procedure how to collect required data for new device?

I'm not familiar with OpenRazer at all, found it only week ago when I bought this keyboard.

donschriefer, 2020-09-10:
I also have an Ornata v2, device isn't recognized :(

pfalomorblab, 2020-09-18:
i've ornata v2 too :( just bought

ddoyle4, 2020-09-25:
Me three

bgoralczyk, 2020-09-25:
I am concerned on how long this might take.  I wish I could help, but this isn't an area that I have experience in.  I might have to return this keyboard and buy a blackwidow instead.

Hellrazor1215, 2020-10-24:
@z3ntu, here are two files I captured. hope they are ok. Never done that before, not even setting up a VM. So if it is not ok or need more or something else do not hesistate to tell me. And also the keyboard had no none effect... so I started from static to another effect.

[ornata.zip](https://github.com/openrazer/openrazer/files/5433899/ornata.zip)

z3ntu, 2020-10-26:
@Hellrazor1215 yep, everything included, thanks! As expected for a Synapse 3 device it doesn't tell us particularly much though unfortunately

Hellrazor1215, 2020-10-26:
@z3ntu I am pretty amaze that I got that on the first try.... if you need help or other infos do not hesitate.

TreyRuffy, 2020-11-28:
I also happen to have the Ornata v2. If you need anything, just DM me and I can try my best to get it to you.

Dev-Owl, 2020-11-28:
Just got one as well, had a look at the mentioned link. Do you still need the files/information? If, I would try to get them in the next days, need to change my setup a bit to follow the steps.

By the way, thanks for the great tooling :+1:

ShimmerGlass, 2020-12-09:
Hello, please find attached a capture of the Ornata V2. I used 3 different effects: fire, wave, and wheel during the capture.

[ornata_v2.pcapng.zip](https://github.com/openrazer/openrazer/files/5668582/ornata_v2.pcapng.zip)

Please tell me if you need anything else!
Thx for your time

RiusmaX, 2020-12-10:
Hi ! 
Thank you for your great work !
I'm also available if you need more information or captures :)

PR #1372 (2021-01-02): Add support for the Ornata Chroma V2
Add support for the Ornata Chroma V2, based on existing keyboards.

What works :
* Effect "none"
* Effect "spectrum"
* Effect "wave"
* Effect "reactive"
* Effect "breath"
* Effect "starlight"
* Effect "static"
* Custom frame write and display
* Luminosity
* Game mode

What doesn't : 
* Effect "ripple" (doesn't seem implemented for other keyboards)

Fixes: #1187
closed 2021-01-02 by z3ntu