openrazer/openrazer #1188: Support for Razer DeathAdder 2000

ShketRybee, 2020-07-06:
### Device Info

* Name: Razer DeathAdder 2000
* Product Number: RZ01-01450200
* Model Number: RZ01-0145

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
Bus 007 Device 002: ID 1532:004f Razer USA, Ltd Razer DeathAdder
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x004f 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
```
Please post the output of
    ls -lah /dev/input/by-id/
```
total 0
drwxr-xr-x 2 root root 120 jul  6 15:09 .
drwxr-xr-x 4 root root 480 jul  6 15:09 ..
lrwxrwxrwx 1 root root   9 jul  6 14:47 usb-Razer_Razer_DeathAdder-event-mouse -> ../event2
lrwxrwxrwx 1 root root   9 jul  6 14:47 usb-Razer_Razer_DeathAdder-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 jul  6 14:47 usb-USB_USB_Keyboard-event-if01 -> ../event4
lrwxrwxrwx 1 root root   9 jul  6 14:47 usb-USB_USB_Keyboard-event-kbd -> ../event3

Packet Captures

Synapse Screenshots
[image]
[image]
[image]

ShketRybee, 2020-07-06:
@z3ntu is captures for this mice required?

z3ntu, 2020-07-07:
The mouse should be pretty similar to #495 hopefully. If it's not too much work then packet captures would be appreciated.

z3ntu, 2020-07-07:
Hm actually your device has the same model & product number as #778 (1532:005F), no clue why they have two different PIDs.

ShketRybee, 2020-07-07:
[deathadder2000.zip](https://github.com/openrazer/openrazer/files/4887391/deathadder2000.zip)

ShketRybee, 2020-07-07:
here is packet captures

z3ntu, 2020-07-11:
@ShketRybee Thanks for the packet captures. I've written a patch for the device but you have to test whether it works properly. The code is on the branch [`feature_deathadder2000_00f4`](https://github.com/openrazer/openrazer/commits/feature_deathadder2000_00f4), instructions for building the package from source is documented at https://github.com/openrazer/openrazer/wiki/Building-a-package .
Do note that the effect selection will not be available in any UI at the moment because it uses new methods for setting those as your device doesn't support colors (just a single color). So please try out the dbus API via e.g. d-feet. If you need any help feel free to ask.

ShketRybee, 2020-07-12:
@z3ntu so i have installed these packages
[image]
but mouse don't work only lightning
dmesg
[ 1373.340336] usb 8-1.1: reset full-speed USB device number 4 using xhci_hcd
[ 1373.552323] usb 8-1.1: reset full-speed USB device number 4 using xhci_hcd
[ 1490.928833] usb 8-1.1: reset full-speed USB device number 4 using xhci_hcd
[ 1491.016427] input: Razer Razer DeathAdder as /devices/pci0000:00/0000:00:04.0/0000:02:00.0/usb8/8-1/8-1.1/8-1.1:1.0/0003:1532:004F.0007/input/input21
[ 1491.016532] hid-generic 0003:1532:004F.0007: input,hidraw3: USB HID v1.11 Mouse [Razer Razer DeathAdder] on usb-0000:02:00.0-1.1/input0
sudo journalctl -b | grep -i razer_mount
июл 12 07:42:37 pompilius razer_mount[3674]: Driver razermouse
июл 12 07:42:37 pompilius razer_mount[3675]: Device_ID 0003:1532:004F.0007
июл 12 07:42:37 pompilius razer_mount[3676]: Unbinding 0003:1532:004F.0007 from hid-generic
июл 12 07:42:37 pompilius razer_mount[3677]: Binding 0003:1532:004F.0007 to razermouse
daemon log
2020-07-12 07:59:21 | razer                          | INFO     | Stopping daemon.
2020-07-12 07:59:23 | razer                          | INFO     | Initialising Daemon (v2.8.0). Pid: 11075
2020-07-12 07:59:23 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2020-07-12 07:59:23 | razer                          | INFO     | Serving DBus
"So please try out the dbus API via e.g. d-feet."
i installed d-feet but don't know how to use this program

z3ntu, 2020-07-12:
Seems fine so far, the device is being bound.

On how to use d-feet:
* Open the application
* Click on the "Session Bus" tab at the top
* Type `razer` into the search
* Click on `org.razer`
* Under `/org/razer` and then `razer.daemon` and `razer.devices` you can find methods that aren't device specific
* Under `/org/razer/device/$serialnumber` you can see your device. If you're not sure what is what you can run `razer.device.misc` -> `getDeviceName()` -> Execute to get the human readable name
* For testing effects go to `razer.device.lighting.logo` and `razer.device.lighting.scroll` and execute the methods there. E.g. for `setLogoStatic` you put e.g. `255, 0, 255` into the "Method input" box and click execute. Then the effect on the device should change.

ShketRybee, 2020-07-12:
@z3ntu there is no /org/razer/device directory and no lighting settings
[image]

z3ntu, 2020-07-13:
Hm weird, can you run `pkill -e openrazer` and then give me the output of `openrazer-daemon -Fv` ? It looks to me that the driver binds correctly on your machine but the daemon doesn't recognize the device for some reason.

ShketRybee, 2020-07-13:
@z3ntu [nautilus@pompilius ~]$ openrazer-daemon -Fv
Starting daemon.
2020-07-13 09:38:42 | razer                          | INFO     | Initialising Daemon (v2.8.0). Pid: 2256
2020-07-13 09:38:42 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus2000 ------------------------ (1532:005E)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEliteDVaEdition ------------- (1532:006A)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEssential ------------------- (1532:006B)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationChroma ------------------ (1532:0F08)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerBasilisk --------------------------- (1532:0064)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2019 --------------------- (1532:0241)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowElite -------------------- (1532:0228)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowEssential ---------------- (1532:0237)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowLite --------------------- (1532:0235)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealth ------------------ (1532:011B)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealthEdition ----------- (1532:010E)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXTournamentEditionChroma - (1532:021A)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018 -------------------------- (1532:0233)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018Base ---------------------- (1532:023B)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018Mercury ------------------- (1532:0240)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerBlade2019Adv ----------------------- (1532:023A)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerBlade2019Base ---------------------- (1532:0246)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerBlade2019StudioEdition ------------- (1532:024D)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerBladeMid2019Mercury ---------------- (1532:0245)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017 ----------------------- (1532:0225)
2020-07-13 09:38:42 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017FullHD ----------------- (1532:022F)
[... cut ...]
2020-07-13 09:38:42 | razer                          | DEBUG    | Adding razer.devices.getOffOnScreensaver method to DBus
2020-07-13 09:38:42 | razer                          | DEBUG    | Adding razer.daemon.version method to DBus
2020-07-13 09:38:42 | razer                          | DEBUG    | Adding razer.devices.syncEffects method to DBus
2020-07-13 09:38:42 | razer                          | DEBUG    | Adding razer.daemon.stop method to DBus
2020-07-13 09:38:42 | razer                          | INFO     | Serving DBus

z3ntu, 2020-07-13:
Weird, anything in `/sys/bus/hid/drivers/razermouse/0*/` ?

ShketRybee, 2020-07-13:
@z3ntu 
[image]
[image]

z3ntu, 2020-07-13:
Okay so the device isn't bound at the moment, but your log before (`sudo journalctl -b | grep -i razer_mount`) said that it was bound at some point. Can you reboot and then check again, and if the device still doesn't appear then give me the output of `LC_ALL=C sudo journalctl -b | grep -i razer_mount` ?

ShketRybee, 2020-07-13:
@z3ntu Jul 13 13:06:04 pompilius razer_mount[969]: Driver razermouse
Jul 13 13:06:04 pompilius razer_mount[970]: Device_ID 0003:1532:004F.0003
Jul 13 13:06:04 pompilius razer_mount[971]: Modprobing razermouse
Jul 13 13:06:04 pompilius razer_mount[979]: Modprobed razermouse
Jul 13 13:06:04 pompilius razer_mount[980]: Unbinding 0003:1532:004F.0003 from hid-generic
Jul 13 13:06:04 pompilius razer_mount[1002]: Binding 0003:1532:004F.0003 to razermouse
the device still doesn't appear in /sys/bus/hid/drivers/razermouse/

z3ntu, 2020-07-13:
Found the issue, please rebuild on the latest commit from the branch. The pkgver should then include 4d94e18

ShketRybee, 2020-07-13:
@z3ntu now mouse moving and every setting in razergenie working fine. Thanks, you can add this commits in the master branch.
[image]

z3ntu, 2020-07-13:
Can you please also test the setLogoStaticMono, setLogoPulsateMono, setScrollStaticMono & setScrollPulsateMono methods with d-feet? These methods should set the scroll wheel or logo led to static or pulsate

ShketRybee, 2020-07-13:
@z3ntu these methods works fine too

closed 2020-09-13 by z3ntu

manoellribeiro, 2020-10-12:
Hello @z3ntu, can you help me with this issue?

Name: Razer DeathAdder 2000
Product Number: RZ01-01450100 (It's the only difference from @ShketRybee)
Model Number: RZ01-0145

I installed open razer in the stable version first, as it didn't work I search for a solution and arrived here, then I followed your instructions, but it didn't help me to get  it working.

Obs: It works well with synapse in windows.

manoellribeiro, 2020-10-17:
@z3ntu yeah, but I tried with the master and it didn't work as well.

z3ntu, 2020-10-18:
@manoellribeiro Can you post the output of `sudo dkms install openrazer-driver/2.8.0` ?

manoellribeiro, 2020-10-18:
@z3ntu 
```
Module openrazer-driver/2.8.0 already installed on kernel 5.4.0-7642-generic/x86_64
```

z3ntu, 2020-10-18:
@manoellribeiro Okay now these commands:
```
$ lsusb | grep 1532
$ sudo modprobe razerkbd
$ ls -al /sys/bus/hid/drivers/razer*/0*/device_serial
```

manoellribeiro, 2020-10-18:
@z3ntu 
```
lsusb | grep 1532:
 Bus 002 Device 002: ID 1532:004d Razer USA, Ltd Razer DeathAdder
sudo modprobe razerkbd:
 nothing
s -al /sys/bus/hid/drivers/razer*/0*/device_serial:
 ls: cannot access '/sys/bus/hid/drivers/razer*/0*/device_serial': No such file or directory
```

z3ntu, 2020-10-18:
@manoellribeiro your device is tracked in https://github.com/openrazer/openrazer/issues/495 , this issue is about 1532:004f

PR #1193 (2020-09-13): Add support for Razer DeathAdder 2000 (1532:004f)
Fixes #1188 

@ShketRybee Best comment here with your test results. Thanks!

- [x] TODO: Add fake driver file