openrazer/openrazer #1190: Razer Firefly v2

wEbAddEr, 2020-07-08:
The Razer Fierfly v2 is not detected.

z3ntu, 2020-07-09:
Can you please share the output of the command `sudo dkms install openrazer-driver/2.8.0` ? Also have you looked at the [help page for Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) at all?

KayJay5, 2020-07-22:
I've got this issue also.

```
:~$ sudo dkms install openrazer-driver/2.8.0
Module openrazer-driver/2.8.0 already installed on kernel 4.15.0-111-generic/x86_64
```

My firefly is being picked up as:
`Bus 003 Device 029: ID 1532:0c04 Razer USA, Ltd `

I've got through the troubleshooting section and everything seems to be working.  I  have a razer mamba and a goliathus extended that are both working.

I've got a Cynosa Chroma Pro that is also not being detected.

Thanks for your help.

z3ntu, 2020-07-22:
1532:0c04 isn't supported yet and 1532:022c isn't either. 

See #946 & #1123 for Firefly and #464 & #1117 for Cynosa Chroma Pro.

Closing this issue then.

closed 2020-07-22 by z3ntu

nightsky30, 2020-07-23:
@KayJay5 Help testing #1117 (specifically ripple effects) would be very much appreciated :)