openrazer/openrazer #1191: BlackWidow Elite: Invalid USB response.

Hendrik-S, 2020-07-10:
### Device Information
Description:    Ubuntu 18.04.4 LTS
Keyboard: BlackWidow Elite
```
journalctl
Jul 10 16:23:52 PC kernel: razer driver: Device data transfer failed.
Jul 10 16:23:52 PC kernel: razer driver: Invalid USB response. USB Report length: -32
Jul 10 16:23:52 PC kernel: razerkbd: Invalid Report Length. Start Marker: 02 id: 1f Num Params: 0c Reserved: 0f Command: 02 Params: 00050800000000000000000000000000 .
```

Might be related to #349 ?

Iridias, 2020-07-14:
I have the same problem (I guess).
I always use static light. But after some time its not changeable anymore, with the same error-messages in dmesg:
```
[ 2154.904032] razer driver: Device data transfer failed.
[ 2154.904240] razer driver: Invalid USB response. USB Report length: -32
[ 2154.904245] razerkbd: Invalid Report Length. Start Marker: 02 id: ff Num Params: 04 Reserved: 00 Command: 81 Params: 01000000000000000000000000000000 .
```

Also a reboot won't help!
Only a re-install seem to make it work again for some time. It doesn't need to be a manual re-install - pacman re-installs the driver every time the kernel is updated.

_Device_: BlackWidow Elite
_OS_: Arch Linux

tbe, 2020-07-21:
Haveing the same issue here, but replugging the device does help for some time. 

Firmware version of my device is v1.0

purple-cmd, 2021-02-10:
Have the same issue, in the same keyboard.

vikasjayaswal, 2021-03-27:
Same Issue btw; This is annoying. All the sudden i can't see the words on my screen!

vikasjayaswal, 2021-04-01:
[    9.180700] kernel: razer driver: Invalid USB response. USB Report length: 26
[    9.180704] kernel: razerkbd: Invalid Report Length. Start Marker: 00 id: 00 Num Params: 00 Reserved: 00 Command: 00 Params: 00000000000000000000000000000000 .
[    9.823816] kernel: iommu ivhd0: AMD-Vi: Event logged [INVALID_DEVICE_REQUEST device=00:00.0 pasid=0x00000 address=0xfffffffdf8000000 flags=0x0a00]

closed 2021-08-31 by github-actions[bot]