openrazer/openrazer #1195: Razer Blackwidow Ultimate Stealth 2014

Taikakim, 2020-07-13:
Could I help somehow in adding support for this keyboard? Now  it's erroneously detected as a Blackwidow Ultimate 2013 and only the static LED strength can be changed.

Taikakim, 2020-07-13:
sudo lsusb -d 1532: -v

Bus 001 Device 002: ID 1532:011a Razer USA, Ltd BlackWidow Ultimate 2013
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x011a BlackWidow Ultimate 2013
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer BlackWidow Ultimate
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)

--
ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 160 heinä  13 23:08 .
drwxr-xr-x 4 root root 320 heinä  13 23:08 ..
lrwxrwxrwx 1 root root   9 heinä  13 23:08 usb-Logitech_USB_Receiver-if01-event-mouse -> ../event9
lrwxrwxrwx 1 root root   9 heinä  13 23:08 usb-Logitech_USB_Receiver-if01-mouse -> ../mouse1
lrwxrwxrwx 1 root root   9 heinä  13 23:08 usb-Razer_Razer_BlackWidow_Ultimate-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 heinä  13 23:08 usb-Razer_Razer_BlackWidow_Ultimate-if01-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 heinä  13 23:08 usb-Razer_Razer_BlackWidow_Ultimate-if02-event-mouse -> ../event6
lrwxrwxrwx 1 root root   9 heinä  13 23:08 usb-Razer_Razer_BlackWidow_Ultimate-if02-mouse -> ../mouse0

Taikakim, 2020-07-13:
The on the fly macro recording does not seem to work. When I press fn+f9, the red light starts to blink, then I press one of the macro buttons and it stays lit until I press it again. Any keys pressed meanwhile don't get recorded, none of the macro keys generate output.

z3ntu, 2020-07-14:
If you can get me [USB packet captures](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol) during Synapse startup I might be able to figure out how Synapse distinguishes between the different devices.

> only the static LED strength can be changed.

What else is possible to be changed?

> The on the fly macro recording does not seem to work

That's possible, we have improvements in open pull requests though which should improve the situation.

ironstarfish, 2021-06-03:
Reviving this issue.  I also have this problem. Cannot run macros recorded. Recording works fine when checking the logs, even execution of the recorded macro is logged correctly, but nothing actually happens.
The log shows the recorded characters being executed but no actual output. M1 shows the macro is recorded and executed on the logs, but still brings up the settings menu when pressed, not the recorded macro.

Captured Synapse startup using USBPcap on Win 10, as well as recording and executing macros.  Let me know if you need anything else.

[RazerBW.zip](https://github.com/openrazer/openrazer/files/6594740/RazerBW.zip)

closed 2021-11-20 by github-actions[bot]