openrazer/openrazer #1198: razor tartarus v2 no keypad key button support

beartravels, 2020-07-14:
ive installed everything under thee sun to get this tto work but to no avial it all does not want to be remaped geni or poymorphus thingy lights fworks ggreat thats not what i bouhgt it for lol
i want to rmap its keys any program in the world for that?
tartus v2
18.04

chabad360, 2020-07-14:
Unfortunately, Openrazer doesn't yet support remapping keys, but it is in the works (#1124). There are tools like keyboarding master or input-modifier that can remap your keys until then.

beartravels, 2020-07-14:
ahh ok ya keyboard master does not work unless youve logged in a super hard wya unles you have a simple trick to log in as root kwyboard maseter is out

beartravels, 2020-07-14:
and su in terminal does not count

chabad360, 2020-07-14:
Keyboard master only needs sudo to install, afterwards you don't need sudo

nova8080, 2020-09-06:
Hey!  I just came across your post while searching around for my own problem.  If you still need to keybind your tartarus v2, it takes a little time, but it is doable. I'll post what worked for me once I finally fix (or give up) my naga trinity mousebutton 4&5 from changing the DPI.

flyer48, 2020-11-04:
Hi nova8080,

yes please tell me how to keybind my tartarus v2, that would be great!

AnTeBiOss, 2020-11-14:
Hi there, i'm Really interested by keybinding on tartarus v2.
How do i help ?

closed 2021-07-22 by github-actions[bot]