openrazer/openrazer #1200: Can not open polychromatic controller

Bryan9995, 2020-07-15:
### Distribution

### Support

- [ ] Have you tried rebooting?
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I can still use the daemon, and change the colors. but the controller doesn't seem to want to launch since the new mint update. I had an issue previously where i had to change the text in a python file (I can't remember which one).

I made a new issue report because it seems like no one else has this issue.

Thanks!

chabad360, 2020-07-16:
This appears to be an issue with [Polycromatic](https://github.com/polychromatic/polychromatic) not OpenRazer. Please close this issue and open one [there](https://github.com/polychromatic/polychromatic/issues).

lah7, 2020-07-16:
It's due to Python 3.8. Be sure to reinstall the package for Ubuntu 20.04 (Linux Mint 20 is based on)

Upgrading from LM 19 will have disabled the PPA. Open **Software Sources**, re-enable under the **PPAs** tab, refresh the Apt cache, and then check for updates again.

If no updates appear after re-enabling the PPA:

    sudo apt install --reinstall polychromatic

https://github.com/polychromatic/polychromatic/issues/245#issuecomment-632052615

closed 2020-07-16 by lah7