openrazer/openrazer #1201: Support for Razer bade 15 2020 advanced (300hz monitor)

TheButlah, 2020-07-17:
### Device Info

* Name: <See under device>
* Product Number: <See under device>
* Model Number: <See under device>

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```

```
Please post the output of
    ls -lah /dev/input/by-id/
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

z3ntu, 2020-07-18:
Yeah, it's a duplicate of #1159 as both devices have the same USB PID, but I'll keep it open as the PR already references it. You can also add a Fixes #1159 to the PR.

TheButlah, 2020-07-19:
Here is a packet capture file, which I recorded after starting razer synapse on windows: https://drive.google.com/file/d/1w2aSdeSqZjMzE4bZPiQJHnARePNA2vTe/view?usp=sharing