openrazer/openrazer #1204: What is the best supported Razer mouse and keyboard by openrazer?

AlbyM-dev, 2020-07-20:
Hello everyone,

I bought a naga trinity mouse but I saw the RGB is still supported only in static mode, this wasn’t mentioned in the documentation though (not complaining).

As per subject, what is the currently most supported ones in terms of features compared to windows?

I am looking at the mid-low range, not super expensive ones.

Thanks for the help

z3ntu, 2020-07-21:
All devices listed at https://openrazer.github.io/ (or in the README) should work fine. The problem with newer devices (the ones only supported in Synapse 3) is that for those Synapse on Windows does a bunch of the heavy lifting as opposed to the firmware on older devices. It should be possible to generate a nice table with hardware effects / features using a few lines of sh but nobody made that yet.
If this answered your question and you don't have any further please close the issue.

AlbyM-dev, 2020-07-25:
Ok thank you so much for the help, appreciated

closed 2020-07-25 by AlbyM-dev