openrazer/openrazer #1205: dkms does not install module for kernel 5.7.8 on Fedora 32

kierun, 2020-07-21:
### Distribution

Fedora 32.

### Support

- [X] Have you tried rebooting?
- [X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

The kernel updated and I generally have to reinstall the module. That is no hassle but this time the module does not install and there are no sensible error message. 

```
√ ; sudo modprobe razerkbd
modprobe: FATAL: Module razerkbd not found in directory /lib/modules/5.7.8-200.fc32.x86_64
✗ 1 ; sudo dkms --verbose install openrazer-driver/2.8.0
√ ; echo $?
0
√  ; sudo journalctl -b | grep -i razer_mount
[…]
1308:Jul 21 09:46:15 nightwatch.neverness.org razer_mount[652]: Modprobing razerkbd
1309:Jul 21 09:46:15 nightwatch.neverness.org systemd-udevd[605]: 0003:1532:0221.0001: Process 'razer_mount razerkbd 0003:1532:0221.0001' failed with exit code 1.
```

Any idea what I am doing wrong?

z3ntu, 2020-07-21:
Can you try running `sudo dkms remove openrazer-driver/2.8.0 --all` and then 
`sudo dkms install openrazer-driver/2.8.0` again?

gaborpapp, 2020-07-21:
I experience the same on Fedora 31, kernel 5.7.8 and 5.7.9.

`dkms remove` worked, but install does not do anything.

```
$ sudo dkms remove openrazer-driver/2.8.0 --all

-------- Uninstall Beginning --------
Module:  openrazer-driver
Version: 2.8.0
Kernel:  5.7.7-100.fc31.x86_64 (x86_64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.
Removing any linked weak-modules

razerkbd.ko.xz:
 - Uninstallation
   - Deleting from: /lib/modules/5.7.7-100.fc31.x86_64/extra/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

razermouse.ko.xz:
 - Uninstallation
   - Deleting from: /lib/modules/5.7.7-100.fc31.x86_64/extra/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

razermousemat.ko.xz:
 - Uninstallation
   - Deleting from: /lib/modules/5.7.7-100.fc31.x86_64/extra/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

razerkraken.ko.xz:
 - Uninstallation
   - Deleting from: /lib/modules/5.7.7-100.fc31.x86_64/extra/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

razeraccessory.ko.xz:
 - Uninstallation
   - Deleting from: /lib/modules/5.7.7-100.fc31.x86_64/extra/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

razercore.ko.xz:
 - Uninstallation
   - Deleting from: /lib/modules/5.7.7-100.fc31.x86_64/extra/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod....

DKMS: uninstall completed.

------------------------------
Deleting module version: 2.8.0
completely from the DKMS tree.
------------------------------
Done.
```

kierun, 2020-07-21:
> Can you try running `sudo dkms remove openrazer-driver/2.8.0 --all` and then
> `sudo dkms install openrazer-driver/2.8.0` again?

`sudo dkms remove openrazer-driver/2.8.0 --all` did remove all the modules. However, I cannot install it again: same behaviour, the command just return 0.

kierun, 2020-07-21:
There was an update to the Fedora 32 kernel and dkms (like just now) and dkms appears to work again. I shall reboot with the new kernel and see if that works. If it does, we have a solution!

kierun, 2020-07-21:
Reboot into kernel ` 5.7.9-200.fc32.x86_64` and dkms has installed the driver just fine. However, we have a new problem:

```
√ ; sudo dkms install openrazer-driver/2.8.0
Module openrazer-driver/2.8.0 already installed on kernel 5.7.9-200.fc32.x86_64/x86_64
√ ; sudo modprobe razerkbd
modprobe: ERROR: could not insert 'razerkbd': Exec format error
✗ 1 ;
```

z3ntu, 2020-07-21:
> Exec format error usually means you are trying to insert module into running kernel which is not the one the module was built for.

I'm trying to upgrade my Fedora VM to 32 at the moment (it's still on 30 and complaining about some things)

kierun, 2020-07-21:
If this help?

```
√ ; sudo dkms install --verbose openrazer-driver/2.8.0 && sudo modprobe razerkbd

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
make -C /lib/modules/5.7.9-200.fc32.x86_64/build M=/var/lib/dkms/openrazer-driver/2.8.0/build clean
make: Entering directory '/usr/src/kernels/5.7.9-200.fc32.x86_64'
make: Leaving directory '/usr/src/kernels/5.7.9-200.fc32.x86_64'

{ KERNELDIR=/lib/modules/5.7.9-200.fc32.x86_64/build make driver; } >> /var/lib/dkms/openrazer-driver/2.8.0/build/make.log 2>&1

make -C /lib/modules/5.7.9-200.fc32.x86_64/build M=/var/lib/dkms/openrazer-driver/2.8.0/build clean
make: Entering directory '/usr/src/kernels/5.7.9-200.fc32.x86_64'
make: Leaving directory '/usr/src/kernels/5.7.9-200.fc32.x86_64'

DKMS: build completed.

razerkbd.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.7.9-200.fc32.x86_64/extra/

razermouse.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.7.9-200.fc32.x86_64/extra/

razermousemat.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.7.9-200.fc32.x86_64/extra/

razerkraken.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.7.9-200.fc32.x86_64/extra/

razeraccessory.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.7.9-200.fc32.x86_64/extra/

razercore.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.7.9-200.fc32.x86_64/extra/
Adding any weak-modules

do_depmod 5.7.9-200.fc32.x86_64

DKMS: install completed.
modprobe: ERROR: could not insert 'razerkbd': Exec format error
✗ 1 ; uname -a
Linux nightwatch.neverness.org 5.7.9-200.fc32.x86_64 #1 SMP Fri Jul 17 16:23:37 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

kierun, 2020-07-21:
I ran `sudo dkms remove openrazer-driver/2.8.0 --all` then `sudo dkms install openrazer-driver/2.8.0`, and rebooted. The module now load fine. I think the previous error was a compile for an old kernel that got installed into the new one.

*The module loads fine with the 3.7.9 kernel and `dkms:2.8`*

gaborpapp, 2020-07-21:
@kierun I'm still having the issue in F31. What's your `dkms` version? It's `dkms-2.8.2-1` in F31, which was updated today as far as I can tell.

kierun, 2020-07-21:
> @kierun I'm still having the issue in F31. What's your `dkms` version? It's `dkms-2.8.2-1` in F31, which was updated today as far as I can tell.

```
√ ; rpm -qa | rg dkms
openrazer-kernel-modules-dkms-2.8.0-2.1.noarch
dkms-2.8.3-1.fc32.noarch
```

Did you upgrade your kernel to 3.7.9?

gaborpapp, 2020-07-21:
Thanks. It seems you have a newer version of `dkms`. I hope it will arrive to F31 soon.

Yes, I'm running the latest kernel, `5.7.9-100.fc31.x86_64`.

z3ntu, 2020-07-21:
With a fresh (and fully updated) Fedora 32 system, the kernel modules build fine with dkms and load fine as well.

kierun, 2020-07-21:
> With a fresh (and fully updated) Fedora 32 system, the kernel modules build fine with dkms and load fine as well.

Was that with kernel 3.7.7 or 3.7.8? What version(s) of dkms?

z3ntu, 2020-07-21:
```
dkms-2.8.3-1.fc32.noarch
openrazer-kernel-modules-dkms-2.8.0-2.1.noarch
kernel-5.6.6-300.fc32.x86_64
```

gaborpapp, 2020-07-21:
I can confirm that it works with `dkms-2.8.3-1` on F31 too (installed if from updates-testing repo as it is still not available in updates).

z3ntu, 2020-07-21:
Lol @ dkms 2.8.3 release notes:

> This release fixes the inability to actually build anything

https://github.com/dell/dkms/releases/tag/v2.8.3

kierun, 2020-07-22:
> Lol @ dkms 2.8.3 release notes:
> 
> > This release fixes the inability to actually build anything
> 
> https://github.com/dell/dkms/releases/tag/v2.8.3

¯\\\_(ツ)\_/¯ Well, I think we have the root cause and can close this!

closed 2020-07-22 by kierun