openrazer/openrazer #1206: Razer Blade Pro 17 2020 support

akaun24, 2020-07-22:
Is there any information on support for blade pro 2020? As of right now daemon is not recognizing anything. Thank you.

z3ntu, 2020-07-23:
Can you please get me the output of the command `lsusb | grep 1532`? This identifies the model that you have, there's likely already an issue open for your device.

akaun24, 2020-07-23:
lsusb | grep 1532
Bus 001 Device 003: ID 1532:0256 Razer USA, Ltd Razer Blade

z3ntu, 2020-07-23:
Okay we don't have an issue for that device yet.

Can you please fill out the template at https://github.com/openrazer/openrazer/issues/new?assignees=&labels=Device+Support&template=new-device-support.md&title=Support+for+%5Bdevice+name+here%5D ?

DmitryLoki, 2020-07-25:
created an issue for Blade 15 Advanced 2020 as well

Bus 001 Device 003: ID 1532:0253 Razer USA, Ltd Razer Blade