openrazer/openrazer #1207: Lighting effects for BlackWidow Ultimate 2013

YamiYukiSenpai, 2020-07-23:
### Distribution

Ubuntu 18.04

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?
Yes
* `$ sudo journalctl -b | grep -i razer_mount`
```
Journal file /var/log/journal/a47e1528d412449c86b74ae3ac1b2093/system@000584cb3978432d-18a66a6c2a386faf.journal~ is truncated, ignoring file.
Jul 20 20:30:49 mc8-Aspire-M3300 razer_mount[557]: Driver razerkbd
Jul 20 20:30:49 mc8-Aspire-M3300 razer_mount[559]: Driver razerkbd
Jul 20 20:30:49 mc8-Aspire-M3300 razer_mount[560]: Device_ID 0003:1532:011A.0002
Jul 20 20:30:49 mc8-Aspire-M3300 razer_mount[558]: Driver razerkbd
Jul 20 20:30:49 mc8-Aspire-M3300 razer_mount[562]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:011A.0002/
Jul 20 20:30:49 mc8-Aspire-M3300 razer_mount[563]: Device_ID 0003:1532:011A.0003
Jul 20 20:30:49 mc8-Aspire-M3300 razer_mount[561]: Device_ID 0003:1532:011A.0001
Jul 20 20:30:49 mc8-Aspire-M3300 razer_mount[565]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:011A.0001/
Jul 20 20:30:49 mc8-Aspire-M3300 razer_mount[567]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:011A.0003/
Jul 20 20:30:49 mc8-Aspire-M3300 razer_mount[569]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:011A.0002/
Jul 20 20:30:49 mc8-Aspire-M3300 razer_mount[570]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:011A.0001/
Jul 20 20:30:49 mc8-Aspire-M3300 razer_mount[571]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:011A.0003/
```
* [Daemon log](https://github.com/openrazer/openrazer/files/4963377/razer_mc8.log)

Can't change the lighting effect of the BlackWidow Ultimate 2013.  I wanted to disable the breathing effect.

z3ntu, 2020-07-23:
> Can't change the lighting effect of the BlackWidow Ultimate 2013.

Can you please elaborate on that? What have you tried? Did you install a GUI like polychromatic or RazerGenie?

YamiYukiSenpai, 2020-07-24:
Yes. Tried both GUIs

z3ntu, 2020-07-24:
The daemon log you posted above doesn't contain any attempts to set an effect

YamiYukiSenpai, 2020-08-07:
That's because there's no option to change the effect

Only color

YamiYukiSenpai, 2020-09-16:
> It seems that polychromatic doesn't support the effects for your device yet.
> Current RazerGenie stable should look like the following where you can change the effect:
> [image]

Guess I'll open an issue at Polychromatic

closed 2021-07-22 by github-actions[bot]