openrazer/openrazer #1208: Support for Razer Blade 15 2020 Base Model

ksnyder21, 2020-07-23:
### Device Info

* Name: Razer Blade 15 2020 Base Model
* Product Number: RZ09-03287E22
* Model Number: RZ09-0328

### Device Information
```
`Bus 001 Device 003: ID 1532:0255 Razer USA, Ltd Razer Blade
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0255 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
`
```

```
`total 0
drwxr-xr-x 2 root root 160 Jul 23 00:15 .
drwxr-xr-x 4 root root 580 Jul 23 00:15 ..
lrwxrwxrwx 1 root root   9 Jul 23 00:15 usb-Azurewave_USB_Camera_0x0001-event-if00 -> ../event3
lrwxrwxrwx 1 root root   9 Jul 23 00:15 usb-Razer_Razer_Blade-event-if01 -> ../event7
lrwxrwxrwx 1 root root   9 Jul 23 00:15 usb-Razer_Razer_Blade-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 Jul 23 00:15 usb-Razer_Razer_Blade-if01-event-kbd -> ../event6
lrwxrwxrwx 1 root root  10 Jul 23 00:15 usb-Razer_Razer_Blade-if02-event-mouse -> ../event10
lrwxrwxrwx 1 root root   9 Jul 23 00:15 usb-Razer_Razer_Blade-if02-mouse -> ../mouse2
`
```

[image]
[image]

_I've never used github (or similar) before so please lemme know if I screwed this up_
Thanks!!!

ksnyder21, 2020-07-23:
the keyboard on this one doesn't have as much lighting features as the advanced model, and the ID's are different than it

nightsky30, 2020-07-23:
This is already covered by #1150 and #1171

z3ntu, 2020-07-23:
Closing as duplicate of #1150 and you can test #1171 as @nightsky30 mentioned

closed 2020-07-23 by z3ntu