openrazer/openrazer #1212: Support for [device name here]

akaun24, 2020-07-25:
### Device Info

* Name: Blade Pro 17 Full HD
* Product Number: RZ09-03295E42
* Model Number: RZ09--0329

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
Bus 001 Device 003: ID 1532:0256 Razer USA, Ltd Razer Blade
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0256 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Blade
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)

```
Please post the output of
    ls -lah /dev/input/by-id/
```
total 0
drwxr-xr-x 2 root root 180 Jul 25 04:34 .
drwxr-xr-x 4 root root 580 Jul 25 04:34 ..
lrwxrwxrwx 1 root root  10 Jul 25 04:34 usb-Azurewave_Integrated_Camera_200901010001-event-if00 -> ../event19
lrwxrwxrwx 1 root root  10 Jul 25 04:34 usb-Azurewave_Integrated_Camera_200901010001-event-if02 -> ../event20
lrwxrwxrwx 1 root root  10 Jul 25 04:34 usb-Razer_Razer_Blade-event-if01 -> ../event12
lrwxrwxrwx 1 root root   9 Jul 25 04:34 usb-Razer_Razer_Blade-event-kbd -> ../event9
lrwxrwxrwx 1 root root  10 Jul 25 04:34 usb-Razer_Razer_Blade-if01-event-kbd -> ../event10
lrwxrwxrwx 1 root root  10 Jul 25 04:34 usb-Razer_Razer_Blade-if02-event-mouse -> ../event16
lrwxrwxrwx 1 root root   9 Jul 25 04:34 usb-Razer_Razer_Blade-if02-mouse -> ../mouse0

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

Baiyah7o, 2020-12-13:
I did a pcap.I hope I did not get it wrong from the wiki page.
[Razer-RZ09-03297G42.zip](https://github.com/openrazer/openrazer/files/5684389/Razer-RZ09-03297G42.zip)

z3ntu, 2020-12-20:
@Baiyah7o Looks good!

decryp2kanon, 2021-03-27:
same model here. how to add?

Xboarder56, 2021-07-14:
Same Model as well except 4k screen. Wondering how we can provide additional logs/resources to get this supported.

ettmetal, 2021-07-23:
> Same Model as well except 4k screen. Wondering how we can provide additional logs/resources to get this supported.

I'd recommend opening a new issue for it using the [New Device](https://github.com/openrazer/openrazer/issues/new?assignees=&labels=Device+Support&template=new-device-support.md&title=Support+for+%5Bdevice+name+here%5D) issue template. It's likeley got a different model number, which will mean it needs its own config. The issue template tells you everything you need to collect :)

z3ntu, 2022-10-15:
Supported since #1376

closed 2022-10-15 by z3ntu