openrazer/openrazer #1213: Help or documentation for installing the driver from source rather than from a deb?

TheButlah, 2020-07-25:
### Distribution

Ubuntu 20.04

### Support

- [X] Have you tried rebooting?
- [X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I've submitted a PR to add some device support, but I'm finding it extremely difficult to figure out how to install my version of the driver + daemon + python library for testing. I've read the troubleshooting guide as well as [this](https://github.com/openrazer/openrazer/wiki/Setting-up-the-keyboard-driver), as well as tried doing all manner of combinations of `sudo make install_i_know_what_im_doing.

The issue is, I don't know what I'm doing. I've managed already to brick my python install after running some of the targets in the makefile, and almost had to reinstall the OS in the process before narrowly fixing it. I remain unable to do a simple 
`python3.8 -c "from openrazer.client import DeviceManager; a = DeviceManager(); print(a.devices)"`, because I can't seem to install the python library.

Can I get a step by step guide on how to set all of this up from source? The makefiles aren't cutting it for me

z3ntu, 2020-07-25:
Does this help? https://github.com/openrazer/openrazer/wiki/Building-a-package

TheButlah, 2020-07-25:
I think that did the trick! It seems to have taken care of binding and unbinding the devices, it lets me echo to the driver to do effects, and `python3.8 -c "from openrazer.client import DeviceManager; a = DeviceManager(); print(a.devices)"` no longer gives an error on importing

TheButlah, 2020-07-25:
I don't know how I missed that before.... I feel rather embarrassed

closed 2020-07-25 by TheButlah