openrazer/openrazer #1215: Support for Razer Basilisk Ultimate

cowboycodur, 2020-07-28:
### Device Info

* Name: Razer Basilisk Ultimate with Charging Dock (https://www.razer.com/gaming-mice/razer-basilisk-ultimate/RZ01-03170100-R3U1)
* Product Number
  * Mouse: RC30-03170100-000
  * Dongle: RC30-03170200-000
  * Dock: RC30-03050200-000
* Model Number
  * Mouse: 
  * Dongle: DGRFG6
  * Dock: RC30-030502

### Device Information
```
Bus 001 Device 011: ID 1532:007e Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x007e 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
[... 170 lines cut ...]
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
```

```
lrwxrwxrwx 1 root root  10 Jul 28 08:05 usb-Razer_Razer_Basilisk_Ultimate_Dongle-event-if01 -> ../event30
lrwxrwxrwx 1 root root  10 Jul 28 08:05 usb-Razer_Razer_Basilisk_Ultimate_Dongle-event-mouse -> ../event27
lrwxrwxrwx 1 root root  10 Jul 28 08:05 usb-Razer_Razer_Basilisk_Ultimate_Dongle-if01-event-kbd -> ../event28
lrwxrwxrwx 1 root root  11 Jul 28 08:05 usb-Razer_Razer_Basilisk_Ultimate_Dongle-if02-event-kbd -> ../event256
lrwxrwxrwx 1 root root   9 Jul 28 08:05 usb-Razer_Razer_Basilisk_Ultimate_Dongle-mouse -> ../mouse3
lrwxrwxrwx 1 root root  10 Jul 28 08:04 usb-RAZER_Razer_Mouse_Dock-event-if01 -> ../event25
lrwxrwxrwx 1 root root  10 Jul 28 08:04 usb-RAZER_Razer_Mouse_Dock-event-mouse -> ../event21
lrwxrwxrwx 1 root root  10 Jul 28 08:04 usb-RAZER_Razer_Mouse_Dock-if01-event-kbd -> ../event22
lrwxrwxrwx 1 root root  10 Jul 28 08:04 usb-RAZER_Razer_Mouse_Dock-if02-event-kbd -> ../event26
lrwxrwxrwx 1 root root   9 Jul 28 08:04 usb-RAZER_Razer_Mouse_Dock-mouse -> ../mouse2
```

jc305697, 2020-09-08:
I can confirm that it doesn't work.

clementinise, 2020-09-23:
Hello!
I would like to contribute to this project, but it seems that I can't give any help since I'm on Windows, am I right?
Would really like to help get support for this mouse and the Firefly V2 mousemat 😊

TimmyOl, 2020-09-27:
The dock is the same but the mouse isn't so its not recognized by the program, probably the same settings as the normal basilisk but with different ID, the difference is that the ultimate is wireless and there is a basilisk hyperspeed but that doesn't have the side button so that is a third option that's not related to this thread 😊

ezXYlay, 2020-09-28:
Same issue here

aRs3N1c42, 2020-10-27:
I too am experiencing this issue.

NerdHarder615, 2020-12-16:
I just got a Basilisk. Is there anything I can do to help get this device supported in open razer? Currently running Manjaro, but have Ubuntu and Fedora systems at home.

z3ntu, 2022-10-15:
This has been fixed with #1321, closing.

closed 2022-10-15 by z3ntu