openrazer/openrazer #1217: Support for Razer Raptor 27 (Monitor)

Aerodin, 2020-07-31:
### Razor Raptor 27

Connects via HDMI/DisplayPort/USB C

Controlled via Razer Synapse via USB C.

Initially I tried setting the RGB values and profile via a friend's borrowed Windows 10 laptop, however, the effects only remain as long as the computer is plugged in via USB C.  Otherwise it switches back to the default of cycling through all colors.

* Name: Razor Raptor 27
* Product Number: <See under device>
* Model Number: RZ39-0276

### Device Information
```
Please post the output of
    lsusb -d 1532: -v

No output information could be seen related to the Raptor 27.
```

```
Please post the output of
    ls -lah /dev/input/by-id/

No output information could be seen related to the Raptor 27.
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

LandingGitHub, 2021-05-08:
Is there any update about the compatibility for the Razer Raptor 27 monitor? I'll would gladly help testing :)

spynin, 2021-06-11:
I also have the same display. Is it more difficult to port due to the usb-c connection utilized for synapse?

hendy643, 2021-06-19:
[Raptor27usbcap.zip](https://github.com/openrazer/openrazer/files/6680608/Raptor27usbcap.zip)
[image]

I've attached a wireshark capture from windows.

I noted the following packets might possibly be related.
The capture seemed to include some kind of device handshake at the start.

59 - 62 toggle underglow on/off
99 - 100 colour to 0%
119 - 120 colour to 10%
135 - 136 colour to 100%
Higher than this: changed colour mode from static to Breathing.

z3ntu, 2021-06-20:
@hendy643 Thanks for the pcap. Basically it looks like any other Razer device, a few things that I see in there:

* Use `razer_chroma_extended_matrix_brightness` with ZERO_LED for brightness
* Use `razer_chroma_extended_matrix_set_custom_frame` for custom frame
* And use `MATRIX_DIMS = [1, 12]` in the daemon

The rest you'll have to guess and try it out whether it works.

Also I haven't seen the USB product id for this device yet, either lsusb on Linux or I guess in the device manager in Windows this should be visible.

LandingGitHub, 2021-06-20:
Nice! some progress :)

jedimasterjonny, 2021-06-23:
> Also I haven't seen the USB product id for this device yet, either lsusb on Linux or I guess in the device manager in Windows this should be visible.

USB PID: `0F12`

jedimasterjonny, 2021-06-28:
To prevent duplication of effort if anyone else comes here - I've picked this up in: https://github.com/openrazer/openrazer/pull/1541

PR #1764 (2022-04-07): Raptor 27 Support
#1541 merge conflicts solved and tested. It works pretty nice for me:

[image]

Closes #1217
closed 2022-04-07 by z3ntu