openrazer/openrazer #1219: 1532:0245 mid-2019 Mercury not recognized

dustinkeib, 2020-08-05:
### Distribution

Mint 20.04 Cinnamon / Ubuntu 20.04 Focal base

### Support

- [x] Have you tried rebooting? 
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Yes, kernel modules built properly, modprobe with no errors

Mid-2019 Razer Blade Mercury 1532:0245 isn't recognized.  Tried on fresh reload of Ubuntu as well as Mint.  Troubleshooting logs below:

**dmesg**
```
% dmesg | grep Razer
[    0.000000] DMI: Razer Blade/CH530, BIOS 1.05 09/19/2019
[    2.509614] usb 1-8: Product: Razer Blade
[    2.509614] usb 1-8: Manufacturer: Razer
[    2.928753] input: Razer Razer Blade as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:1532:0245.0006/input/input12
[    2.986250] hid-generic 0003:1532:0245.0006: input,hidraw5: USB HID v1.11 Keyboard [Razer Razer Blade] on usb-0000:00:14.0-8/input0
[    2.987178] input: Razer Razer Blade Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:1532:0245.0007/input/input13
[    3.046660] input: Razer Razer Blade Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:1532:0245.0007/input/input14
[    3.046806] input: Razer Razer Blade System Control as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:1532:0245.0007/input/input15
[    3.046926] input: Razer Razer Blade as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:1532:0245.0007/input/input16
[    3.047122] hid-generic 0003:1532:0245.0007: input,hidraw6: USB HID v1.11 Keyboard [Razer Razer Blade] on usb-0000:00:14.0-8/input1
[    3.047671] input: Razer Razer Blade as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:0245.0008/input/input17
[    3.047816] hid-generic 0003:1532:0245.0008: input,hidraw7: USB HID v1.11 Mouse [Razer Razer Blade] on usb-0000:00:14.0-8/input2
```

**lsusb**
```
%  lsusb -d 1532: -v                                                                                                                                                                                                               
                                                                                                                                                                                                                                              
Bus 001 Device 005: ID 1532:0245 Razer USA, Ltd Razer Blade                                                                                                                                                                                   
Couldn't open device, some information will be missing                                                                                                                                                                                        
Device Descriptor:                                                                                                                                                                                                                            
  bLength                18                                                                                                                                                                                                                   
  bDescriptorType         1                                                                                                                                                                                                                   
  bcdUSB               2.00                                                                                                                                                                                                                   
  bDeviceClass            0                                                                                                                                                                                                                   
  bDeviceSubClass         0                                                                                                                                                                                                                   
  bDeviceProtocol         0                                                                                                                                                                                                                   
  bMaxPacketSize0        64                                                                                                                                                                                                                   
  idVendor           0x1532 Razer USA, Ltd                                                                                                                                                                                                    
  idProduct          0x0245                                                                                                                                                                                                                   
  bcdDevice            2.00                                                                                                                                                                                                                   
  iManufacturer           1                                                                                                                                                                                                                   
  iProduct                2                                                                                                                                                                                                                   
  iSerial                 0                                                                                                                                                                                                                   
  bNumConfigurations      1                                                                                                                                                                                                                   
  Configuration Descriptor:                                                                                                                                                                                                                   
    bLength                 9                                                                                                                                                                                                                 
    bDescriptorType         2                                                                                                                                                                                                                 
    wTotalLength       0x0054                                                                                                                                                                                                                 
    bNumInterfaces          3                                                                                                                                                                                                                 
    bConfigurationValue     1                                                                                                                                                                                                                 
    iConfiguration          0                                                                                                                                                                                                                 
    bmAttributes         0xa0                                                                                                                                                                                                                 
      (Bus Powered)                                                                                                                                                                                                                [... cut ...]
lrwxrwxrwx 1 root root 0 Jul 27 22:39 0018:04F3:30A6.0001 -> ../../../devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-2/i2c-ELAN0406:00/0018:04F3:30A6.0001
```
**modprobe**
```
% modprobe razerkbd
[dustin:~] % lsmod | grep razerkbd
razerkbd               61440  0
hid                   131072  5 i2c_hid,usbhid,hid_multitouch,razerkbd,hid_generic
```

Sorry for the info dump :)  I couldn't find a related issue to +1, so please point me in the direction if there is one.  Thank you for making this available and please let me know if I can provide any further information.

Thanks!
Dustin

lah7, 2020-08-05:
Hi, what's the output of the daemon?

    openrazer-daemon -Fv

Has your user account been added to the `plugdev` group?

    groups
    sudo gpasswd -a plugdev $USER

I suspect this could be why the applications won't recognise the hardware. Once added, you will need to restart the session/computer.

dustinkeib, 2020-08-06:
Hi, thanks for the help!  

```
θ64° [dustin:~] % groups                                                                                                                                                                                                                 
dustin adm cdrom sudo dip plugdev lpadmin sambashare docker                                                                                                                                                                       
```

```
θ64° [dustin:~] % pkill -e openrazer && openrazer-daemon -Fv                                                                                                                                                                                  
openrazer-daemo killed (pid 3584)                                                                                                                                                                                                             
Starting daemon.                                                                                                                                                                                                                              
2020-08-05 18:51:13 | razer                          | INFO     | Initialising Daemon (v2.5.0). Pid: 22015                                                                                                                                    
2020-08-05 18:51:13 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor                                                                                                                                       
2020-08-05 18:51:13 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)                                                                                          
2020-08-05 18:51:13 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)                                                                                          
2020-08-05 18:51:13 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus2000 ------------------------ (1532:005E)                                                                                          
2020-08-05 18:51:13 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)                                                                                          
2020-08-05 18:51:13 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)                                                                                          
2020-08-05 18:51:13 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)                                                                                          
2020-08-05 18:51:13 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)                                                                                          
2020-08-05 18:51:13 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)                                                                                          
2020-08-05 18:51:13 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)                                                                                          
2020-08-05 18:51:13 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealth ------------------ (1532:011B)                                                                                          
2020-08-05 18:51:13 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealthEdition ----------- (1532:010E)                                                                                          
2020-08-05 18:51:13 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)                                                                                          
2020-08-05 18:51:13 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)                                                                                          
2020-08-05 18:51:13 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)                                                                                          
2020-08-05 18:51:13 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)                                                                                          
2020-08-05 18:51:13 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXTournamentEditionChroma - (1532:021A)                                                                       [... cut ...]
2020-08-05 18:51:13 | razer                          | DEBUG    | Adding razer.devices.getSyncEffects method to DBus
2020-08-05 18:51:13 | razer                          | DEBUG    | Adding razer.devices.supportedDevices method to DBus
2020-08-05 18:51:13 | razer                          | DEBUG    | Adding razer.devices.syncEffects method to DBus
2020-08-05 18:51:13 | razer                          | INFO     | Serving DBus
2020-08-05 18:51:35 | razer                          | DEBUG    | DBus called get_serial_list
```

Thanks, I'm sure there's something I've missed!

lah7, 2020-08-06:
Ah, it looks like you've got an older version installed (v2.5.0) from the [Ubuntu/Debian archives](https://launchpad.net/openrazer/+packages), but your device was added in v2.6.0. [The latest is currently v2.8.0.](https://github.com/openrazer/openrazer/releases/latest)

Per [the instructions for Ubuntu on the homepage](https://openrazer.github.io/#ubuntu), add the OpenRazer "stable" PPA, and then update the system to acquire the latest packages. :slightly_smiling_face: 

    sudo add-apt-repository ppa:openrazer/stable
    sudo apt dist-upgrade                         # or open "Software Updater"

dustinkeib, 2020-08-15:
Thank you and sorry for the delay in getting back to you on this.  It works!

closed 2020-08-15 by dustinkeib