openrazer/openrazer #1221: Requesting Support for the Razer Blade Stealth Mercury White (2020 edition)

Daurnatha, 2020-08-06:
### Distribution

Razer Blade Stealth 2020 - 10th gen i7, intel iris plus graphics
### Support
Hi guys, im very new to the community/linux as a whole, ive installed open razer and polychromatic but seen that this model doesnt have support yet and was hoping someone could help.
- [ ] Have you tried rebooting?
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

eudy97, 2020-09-06:
> Hi, can you post the output of `lsusb | grep 1532`

Bus 003 Device 003: ID 1532:0252 Razer USA, Ltd Razer Blade

z3ntu, 2020-09-06:
1532:0252 is already supported and will be a part of the next release.

closed 2020-09-06 by z3ntu