openrazer/openrazer #1223: Daemon Issue?

LuxLucian, 2020-08-07:
I am running Solus 4.1 and have bumped into this issue. RazerGenie gives me an error message (Polychromatic does the same) and I am unsure what's wrong. I have a Razer Cynosa Chroma and a Mamba Wireless.
[image]
[image]

LuxLucian, 2020-08-10:
https://pastebin.com/RUUxHTv1
https://pastebin.com/GWn82443
[image]

[image]

chabad360, 2020-08-11:
It appears that Solus doesn't support DKMS, so your installation process will be a bit more manual. This may be of use to you: https://discuss.getsol.us/d/1222-how-to-install-github-sources-requiring-dkms-for-missing-devices-drivers/8

LuxLucian, 2020-08-11:
Just to clarify, would I DL the source code for openrazer extract it to a temp directory, and compile the items? This would install it properly, assuming the compile succeeds? Sorry, bit new to compile from source, would rather not break something.

chabad360, 2020-08-11:
Yup!

LuxLucian, 2020-08-11:
So, I successfully compiled the source code, but when I try to install razergenie I'm still prompted to install openrazer. Should I do that or am I missing an important step?

z3ntu, 2020-08-11:
I'd suggest you contact the maintainers of the package at Solus - it should "just work" when installed as that's how they provide the package.

closed 2020-08-11 by z3ntu