openrazer/openrazer #1228: Razer Turret For Xbox One (incl. Gears Of War Edition)

aSqueaker, 2020-08-16:
### Device Info

* Name: Razer Turret for Xbox One
* Product Number: R284-02820300
* Model Number: R203-0282

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
Bus 003 Device 016: ID 1532:0a21 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0a21 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
[... 208 lines cut ...]
```
Please post the output of
    ls -lah /dev/input/by-id/
```
lrwxrwxrwx 1 root root   11 Aug 16 18:33 usb-Razer_Razer_Turret_Keyboard_Xbox_One_Edition-event-if01 -> ../event262
lrwxrwxrwx 1 root root   11 Aug 16 18:33 usb-Razer_Razer_Turret_Keyboard_Xbox_One_Edition-event-if03 -> ../event264
lrwxrwxrwx 1 root root   11 Aug 16 18:33 usb-Razer_Razer_Turret_Keyboard_Xbox_One_Edition-event-kbd -> ../event258
lrwxrwxrwx 1 root root   11 Aug 16 18:33 usb-Razer_Razer_Turret_Keyboard_Xbox_One_Edition-if01-event-kbd -> ../event259
lrwxrwxrwx 1 root root   11 Aug 16 18:33 usb-Razer_Razer_Turret_Keyboard_Xbox_One_Edition-if02-event-mouse -> ../event263
lrwxrwxrwx 1 root root    9 Aug 16 18:33 usb-Razer_Razer_Turret_Keyboard_Xbox_One_Edition-if02-mouse -> ../mouse3
lrwxrwxrwx 1 root root    9 Aug 16 18:11 usb-Razer_Razer_Turret_Receiver-event-if01 -> ../event6
lrwxrwxrwx 1 root root    9 Aug 16 18:11 usb-Razer_Razer_Turret_Receiver-event-if02 -> ../event8
lrwxrwxrwx 1 root root    9 Aug 16 18:11 usb-Razer_Razer_Turret_Receiver-event-mouse -> ../event3
lrwxrwxrwx 1 root root    9 Aug 16 18:11 usb-Razer_Razer_Turret_Receiver-if01-event-kbd -> ../event4
lrwxrwxrwx 1 root root    9 Aug 16 18:11 usb-Razer_Razer_Turret_Receiver-mouse -> ../mouse0

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

(_REALLY_ hoping you know how to fix the "Xbox" key on this otherwise beautifully-designed keyboard.  They put it in place of the right [Control] key, and I did test it works in Windows (as a Control key), however, I use Ubuntu, where it is completely non-functional.  This usability hindrance is all I care about, not the lighting stuff.

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

quinndiggity, 2020-09-15:
duplicate of #1060 ; I haven't had a chance to do the packet capture from mine - @tkeinan / @aSqueaker if either of you beat me to it it'd be bomb.com https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol#phase-3---reverse-engineering

z3ntu, 2020-09-16:
@quinndiggity #1060 has a different USB PID (0x0904) and not 0x0a21 / 0x0a23

free-vbucks, 2020-09-28:
Here are the pictures in Synapse **for the mouse** (the Razer Turret is two devices using the same USB dongle, but each device can be used independently on the same dongle *or* independently while wired)
Top toolbar-
[image]
Customize-
[image]
Performance-
[image]
Lighting-
[image]
Calibration-
[image]
Power-
[image]

free-vbucks, 2020-09-28:
Sorry meant to comment that in #1060