openrazer/openrazer #1230: Support for Huntsman Mini (60%)

Cipulot, 2020-08-18:
### Device Info

* Name: Razer Huntsman Mini
* Model Number: RZ03-03390500-R3U1

### Device Information
```
Bus 001 Device 003: ID 1532:0257 Razer USA, Ltd Razer Huntsman Mini
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0257 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x006d
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     177
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0016  1x 22 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              2 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     348
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval               1
```

```
total 0
drwxr-xr-x 2 root root 240 Aug 18  2020 .
drwxr-xr-x 4 root root 920 Aug 18  2020 ..
lrwxrwxrwx 1 root root  10 Aug 18  2020 usb-Chicony_Electronics_Co._Ltd._HP_Wide_Vision_FHD_Camera_1-event-if00 -> ../event21
lrwxrwxrwx 1 root root  10 Aug 18  2020 usb-Chicony_Electronics_Co._Ltd._HP_Wide_Vision_FHD_Camera_1-event-if02 -> ../event22
lrwxrwxrwx 1 root root  10 Aug 18  2020 usb-Logitech_USB_Receiver-if02-event-mouse -> ../event16
lrwxrwxrwx 1 root root   9 Aug 18  2020 usb-Logitech_USB_Receiver-if02-mouse -> ../mouse5
lrwxrwxrwx 1 root root   9 Aug 18  2020 usb-Razer_Razer_Huntsman_Mini_00000000001A-event-if01 -> ../event8
lrwxrwxrwx 1 root root  10 Aug 18  2020 usb-Razer_Razer_Huntsman_Mini_00000000001A-event-if03 -> ../event13
lrwxrwxrwx 1 root root   9 Aug 18  2020 usb-Razer_Razer_Huntsman_Mini_00000000001A-event-kbd -> ../event6
lrwxrwxrwx 1 root root   9 Aug 18  2020 usb-Razer_Razer_Huntsman_Mini_00000000001A-if01-event-kbd -> ../event7
lrwxrwxrwx 1 root root  10 Aug 18  2020 usb-Razer_Razer_Huntsman_Mini_00000000001A-if02-event-mouse -> ../event12
lrwxrwxrwx 1 root root   9 Aug 18  2020 usb-Razer_Razer_Huntsman_Mini_00000000001A-if02-mouse -> ../mouse1

```

jnreynoso, 2020-11-04:
Waiting for this .... :+1:

Cipulot, 2020-11-04:
@jnreynoso In the next week I'll manage to solve the pcap needs for the issue. Didn't managed to get it properly for a while. After that support should be implemented in a future release (I hope... some older devices are still not supported)

jnreynoso, 2020-11-04:
Great, thanks @Cipulot , I'm looking forward to the release ...

Cipulot, 2020-11-09:
Little update
I'm having some problems with the pcap thing. It seems that the capture turn out to be corrupted every time I save the file.

In the meantime, if anyone have the Huntsman Mini can share their pcap here.

nikolastojkovic1, 2020-11-11:
Hello @Cipulot ,
I have Huntsman Mini and I would be happy to try to share pcap.
Since I haven't tried doing that before, can you share some guide on capturing pcap?

Cipulot, 2020-11-11:
Hello there @nikolastojkovic1 ,
I've followed this guide: https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol
It was liked by z3ntu in this issue #1261

Joseos123, 2020-11-29:
I would also provide any and all help you need because this driver would be amazing. Fn key likes to stop toggling hypershift mode so none of my function or media keys work :/

MasterEnoc, 2020-12-04:
Hi everyone, I have a huntsman mini and I have access to a windows pc. can I use wireshark with usbcap in windows (without being in a vm) for reverse engineer purposes?

z3ntu, 2020-12-20:
@MasterEnoc I've heard that this works as well - doesn't hurt to try at least.

JakeQuilty, 2020-12-22:
Hey @Cipulot, are you still working on this? Do you have a work-in-progress branch I could take a look at? I'd like to help if I can because I need this driver too.

Cipulot, 2020-12-23:
Hi @JakeQuilty, unfortunately I don't have a work-in-progress branch at the moment:(
Also my main pc is acting strange with usb devices in general lately, so I don't think that the produced captures are valid.

elliott-maguire, 2021-01-19:
Also looking for Huntsman Mini support - if there's anything I could do to help, let me know.

ksonne, 2021-01-20:
@elliott-maguire same here! The lighting on my 2020 Blade 15 works like a charm! Would be a huge win to sync it with my Huntsman Mini! Currently it just cycles colors ;) I'll try to do some pcap, but I've never done this before. If I find out anything useful, I'll let you know. Cheers!

ernestochero, 2021-02-24:
Hi everyone, is there any update on this release?

ne04, 2021-03-20:
How's progress coming along on this? Would love support for this keyboard :)

closed 2021-03-21 by z3ntu

bphd, 2022-06-17:
What about the impossibility to perform `Control` + `Alt` + `Fi`/`Delete`? Because the keyboard is confused between the `Function` key to triggers certain actions and `Control` to change keyboard color scheme? On Linux, it is essential to be able to switch between virtual terminals

PR #1431 (2021-03-21): Add support for keyboard: Huntsman Mini
Add support requested in #1230. I tested the driver following the wiki instructions, I will build  and install the .deb package to test everything work as intended. 

Currently two effects are not supported by openrazer:
- Fire
- Wheel

I will try to add this effects, but it can take me a while since I do not understand the openrazer source code very well. Images and pcapng files needed by this driver are attached below.

[wiresharkRazer.zip](https://github.com/openrazer/openrazer/files/5945193/wiresharkRazer.zip)

Fixes #1230