openrazer/openrazer #1231: Add OpenRazer to systemd startup

Lockna, 2020-08-18:
### Distribution
Arch Linux 
Kernel Version: 5.8.1
### Support

- [ x] Have you tried rebooting?
- [ x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I was wondering how I can add the openrazer-daemon to the systemd startup.
I have to start it manually or it starts when I open the polychromatic frontend.
When I type in 
`sudo systemctl enable openrazer-daemon`
it gives me this output
`Failed to enable unit: Unit file openrazer-daemon.service does not exist.`
Am I missing something?

z3ntu, 2020-08-18:
`systemctl --user enable openrazer-daemon.service` should work for that.

closed 2020-08-18 by z3ntu

Lockna, 2020-08-18:
Nope, doesn't work either
`Failed to connect to bus: No such file or directory`