openrazer/openrazer #1232: Support for Razer Naga Left-Handed Edition (2020 version)

queer, 2020-08-19:
### Device Info

* Name: Razer Naga Left-Handed Edition
* Product Number: RZ0103410100
* Model Number: RZ01-0341

If necessary, [here is a link to it on the Razer website](https://www.razer.com/gaming-mice/razer-naga-left-handed-edition/RZ01-03410100-R3M1).

### Device Information
```
Bus 005 Device 004: ID 1532:008d Razer USA, Ltd Razer Naga Left Handed Edition
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x008d 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Naga Left Handed Edition
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x006d
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     325
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

```
total 0
drwxr-xr-x 2 root root 280 Aug 19 12:27 ./
drwxr-xr-x 4 root root 740 Aug 19 12:27 ../
lrwxrwxrwx 1 root root  10 Aug 18 20:20 usb-Razer_Razer_Naga_Left_Handed_Edition-event-if01 -> ../event14
lrwxrwxrwx 1 root root   9 Aug 18 20:20 usb-Razer_Razer_Naga_Left_Handed_Edition-event-mouse -> ../event8
lrwxrwxrwx 1 root root   9 Aug 18 20:20 usb-Razer_Razer_Naga_Left_Handed_Edition-if01-event-kbd -> ../event9
lrwxrwxrwx 1 root root  10 Aug 18 20:20 usb-Razer_Razer_Naga_Left_Handed_Edition-if02-event-kbd -> ../event26
lrwxrwxrwx 1 root root   9 Aug 18 20:20 usb-Razer_Razer_Naga_Left_Handed_Edition-mouse -> ../mouse1
lrwxrwxrwx 1 root root  10 Aug 18 20:20 usb-Wacom_Co._Ltd._Cintiq_16_9BQ00X1004064-event-mouse -> ../event28
lrwxrwxrwx 1 root root   9 Aug 18 20:20 usb-Wacom_Co._Ltd._Cintiq_16_9BQ00X1004064-mouse -> ../mouse2
lrwxrwxrwx 1 root root   9 Aug 18 20:20 usb-ZSA_Ergodox_EZ_0-event-if01 -> ../event6
lrwxrwxrwx 1 root root   9 Aug 18 20:20 usb-ZSA_Ergodox_EZ_0-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 Aug 18 20:20 usb-ZSA_Ergodox_EZ_0-if01-event-kbd -> ../event7
lrwxrwxrwx 1 root root   9 Aug 18 20:20 usb-ZSA_Ergodox_EZ_0-if01-event-mouse -> ../event4
lrwxrwxrwx 1 root root   9 Aug 18 20:20 usb-ZSA_Ergodox_EZ_0-if01-mouse -> ../mouse0
```

### Packet Captures / Synapse Screenshots

I unfortunately do not own a Windows machine, so I'm not sure how I could provide these. I am willing to donate the hardware if needed.

hendy643, 2020-08-19:
They're not the best, but hopefully, they can help.

[screens.zip](https://github.com/openrazer/openrazer/files/5099336/screens.zip)
[cap.zip](https://github.com/openrazer/openrazer/files/5099337/cap.zip)

MarkJBurton, 2020-11-27:
I also have the same Razer Naga Left-Handed Edition (2020 version)
Hope it gets added sometime.

z3ntu, 2020-11-29:
From the pcap I can tell you that this device uses `razer_chroma_extended_matrix_effect_custom_frame` with transaction id 0x1f and has matrix dimensions of [6, 22]. Additionally it does normal dpi (like most devices) and supported the dpi stages feature. 
I hope somebody can start implementing support for the device with this info.

cylgalad, 2020-12-19:
> From the pcap I can tell you that this device uses `razer_chroma_extended_matrix_effect_custom_frame` with transaction id 0x1f and has matrix dimensions of [6, 22]. Additionally it does normal dpi [...]

Matrix is [1, 3] it's a :computer_mouse: not a :keyboard: :wink:

z3ntu, 2020-12-20:
Right... There's something in the pcaps with `[6, 22]` though, but also `[1, 3]` as you've correctly stated. Maybe a keyboard was also connect during the captures. :shrug:

hendy643, 2021-01-27:
PR #1414

PR #1414 (2021-01-31): Support for Naga Lefthanded edition 2020.
Added support for logo lighting and scroll wheel lighting for spectrum/wave/reactive/static and breath for scroll wheel.  Brightness for scroll wheel and logo. DPI support, and polling frequency.

Not done: lighting or mappings for side buttons.

Bugs:
BUG: Device 0003:1532:008D.000C has already been found with interface 0003:1532:008D.0009. Skipping
ran formatting script.

Patch from cylgalad incorporated.

https://github.com/openrazer/openrazer/pull/1361#issuecomment-748494191
formatted files

merge master into feature branch

ran ./scripts/generate_ [... cut]
closed 2021-01-31 by z3ntu