openrazer/openrazer #1234: Razer id/password

eldiener, 2020-08-24:
### Distribution

### Support

- [ ] Have you tried rebooting?
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

How do I provide my Razer id/password to Openrazer so that it will use my cloud-based keyboard settings under Linux ?

parkerlreed, 2020-08-24:
You don't. This isn't a thing. This is a basic kernel driver/userland controller with some frontend applications. All the profiles and settings are unique to whatever frontend you use.

eldiener, 2020-08-24:
I thought this would enable me to share my Razer Synapse settings so that when I used a Linux OS with openrazer it would pick up those settings and use it. If it can not do that I personally have no use for it, since I have no inclination to have to duplicate all Razer settings on each OS. Thanks for the help and good luck !

z3ntu, 2020-08-24:
We don't have any cloud connectivity at all due to several reasons, but mostly privacy and probably the lack of public API from Razer.

closed 2020-08-24 by z3ntu

eldiener, 2020-08-26:
Privacy should not be an issue for cloud connectivity since it would be the end-user's choice whether or not his/her settings can be accessed or not when he/she gives you the id/password, which can be encrypted/decrypted easily enough on whatever OS it is used. If there is no public API however for accessing the cloud-based settings then there is nothing you can do. But if there is an API for doing so, along with well-defined information of what format the cloud-based information is in, then there is no reason you could not provide that functionality in openrazer. Of course I fully understand that if it has never been requested you would not have any impetus to do it, but I am surprised that no others but me assumed that it must be part of openrazer to be useful to them. The Razer website certainly does not help with any findable information about a Razer API/SDK, so maybe this does not exist at all.

z3ntu, 2020-08-26:
If you want to implement this, go ahead, the project is open-source. But I definitely don't have the need for such integration and as I said privacy is a big aspect of it.

lah7, 2020-08-26:
If Razer Synapse exposes the save data files on the local computer, then it **may** be possible to create a script for this repository to convert effects to OpenRazer's Python library (API). It won't be easy for other features (like profiles and key remapping) as they don't currently have an equivalent at an OpenRazer level.

eldiener, 2020-08-26:
I was told that a user's Razer data is cloud-based. If the Razer SDK had functionality to retrieve or update this data based on the id/password then openrazer could do it also. I am not saying that openrazer would want to do it, but I am just saying that it might be possible. So far, looking at information on the Razer Insider web site, Razer buries any information about their SDK so throughly, I am surprised openrazer can do anything at all based on the Razer SDK <g>. But my own need as a user is definitely to be abl;e to use my Razer information in a cross-platform manner without having to duplicate it on each platform into which I boot. I am a bit surprised that no one else has brought this need up before, but each to his own needs.