openrazer/openrazer #1236: Support for Razer Blade Stealth 2020

ellie, 2020-08-25:
### Device Info

* Name: Razer Blade Stealth 2020
* Product Number: RZ09-03102W52
* Model Number: RZ09-0310

### Device Information
```
Bus 003 Device 004: ID 1532:0252 Razer USA, Ltd Razer Blade
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0252 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

```

```
usb-Razer_Razer_Blade-event-if01 ⇒ ../event8
usb-Razer_Razer_Blade-event-kbd ⇒ ../event5
usb-Razer_Razer_Blade-if01-event-kbd ⇒ ../event6
usb-Razer_Razer_Blade-if02-event-mouse ⇒ ../event10
usb-Razer_Razer_Blade-if02-mouse ⇒ ../mouse2
```
I'd be pretty interested in helping out here, but I've never done this sort of thing before - is there anything else I should read as well as https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol?

Thanks!

lah7, 2020-08-26:
I'm far from being proficient in C or the driver code, but looking at other Blade pull requests as a model example to add support for yours should be useful:

> Add support for Razer Blade Stealth Late 2019
> https://github.com/openrazer/openrazer/pull/962/files

There's a good chance supporting your model is a matter of copy and pasting, adapting the IDs/names to match your model, and compiling/modprobing the driver. If you're using Ubuntu/Debian, you can use `./scripts/create-debs.sh` to create packages for local testing. :ok_hand: 

Getting pcaps and inspecting the USB protocol would be necessary if your model uses a different/newer protocol and needs further research.

ellie, 2020-08-26:
Oh perfect, that makes a lot of sense - I'll have a go sometime in the next week or so then, thank you!

durzo, 2020-09-22:
I'm using the latest daily release and my 2020 is supported just fine. looks like it was added back in June

z3ntu, 2020-09-26:
Yep, device is already supported.

closed 2020-09-26 by z3ntu