openrazer/openrazer #1237: Cynosa Lite not recognized

cursorme, 2020-08-25:
### Distribution

manjaro 20

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Hi,

I've follow all steps and on my manjaro and all setup looks good. Daemon is working just fine but my Cynosa Lite does not get recognized.
My info is the following:
[ebarbosa@embbmanj01 ~]$ tail ~/.local/share/openrazer/logs/razer.log
2020-08-25 18:50:38 | razer | INFO | Initialising Daemon (v2.8.0). Pid: 9499
2020-08-25 18:50:38 | razer.screensaver | INFO | Initialising DBus Screensaver Monitor
2020-08-25 18:50:38 | razer | INFO | Serving DBus
2020-08-25 19:03:08 | razer | INFO | Initialising Daemon (v2.8.0). Pid: 8155
2020-08-25 19:03:08 | razer.screensaver | INFO | Initialising DBus Screensaver Monitor
2020-08-25 19:03:08 | razer | INFO | Serving DBus
2020-08-25 19:03:13 | razer | INFO | Stopping daemon.
2020-08-25 19:03:14 | razer | INFO | Initialising Daemon (v2.8.0). Pid: 8291
2020-08-25 19:03:14 | razer.screensaver | INFO | Initialising DBus Screensaver Monitor
2020-08-25 19:03:14 | razer | INFO | Serving DBus
[ebarbosa@embbmanj01 ~]$ sudo modprobe razerkbd
[ebarbosa@embbmanj01 ~]$ sudo dkms install openrazer-driver/2.8.0
Module openrazer-driver/2.8.0 already installed on kernel 5.8.1-3-MANJARO/x86_64
[ebarbosa@embbmanj01 ~]$ lsusb | grep '1532:'
Bus 001 Device 005: ID 1532:023f Razer USA, Ltd RZ03-0274 Gaming Keyboard [Cynosa Lite]
[ebarbosa@embbmanj01 ~]$

What am I missing ?

Thanks
Eduardo

z3ntu, 2020-08-25:
Your device isn't in a stable openrazer release yet, you can use openrazer-git from the AUR in the meantime. Does that solve your question?

cursorme, 2020-08-26:
Thanks for the feedback. Do you mean https://aur.archlinux.org/packages/openrazer-meta-git/ ? 
I've try that one, also run modprobe after + dkms install just to make sure but everything is in the same situation.
But on device support list I can see my device: https://github.com/openrazer/openrazer#device-support

z3ntu, 2020-08-26:
> Do you mean https://aur.archlinux.org/packages/openrazer-meta-git/ ?

Yes (the pkgbase is called openrazer-git - https://aur.archlinux.org/pkgbase/openrazer-git/ but it doesn't matter)

> But on device support list I can see my device: https://github.com/openrazer/openrazer#device-support

The readme is from git master where your device is supported - but you tried the latest release 2.8.0 which didn't have your device included yet.

cursorme, 2020-08-26:
Makes sense and thanks for the feedback.
Seems that then might be an issue with Cynosa Lite on master then.
What is the best way to report this as an bug , this ticket will work ?

z3ntu, 2020-08-26:
Just write it here for now, you can always move it to a new report if it indeed is a major bug.

Also fyi most of the time it's less work for the maintainers if you just do something rather than ask to do something (i.e. in this case you could've just written your issue here and it's always possible to move it to a new ticket - rather than ask which takes some time to respond to :wink: )

Guitlle, 2020-11-08:
So this keyboard is not yet supported on the packaged version in the PPA, How can I install openrazer from source to get this working? I didn't find any instructions to build from source and install properly. I dit the make install_i_know_what_im_doing but the keyboard didn't work anymore until I uninstalled.

z3ntu, 2020-11-08:
@Guitlle 

> I dit the make install_i_know_what_im_doing

Maybe you didn't know what you're doing? :thinking: 

Anyways you can follow https://github.com/openrazer/openrazer/wiki/Building-a-package which has instructions for most distros.

closed 2021-06-20 by github-actions[bot]