openrazer/openrazer #1238: Support for Razer Pro Type keyboard

R-Bell, 2020-08-27:
### Device Info

* Name: Razer Pro Type
* Product Number: RZ03-03070100
* Model Number: RC30-030702

### Device Information
```
Bus 003 Device 004: ID 1532:0249 Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0249 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Pro Type
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     179
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

```
usb-Razer_Razer_Pro_Type-event-if01 -> ../event11
usb-Razer_Razer_Pro_Type-event-mouse -> ../event8
usb-Razer_Razer_Pro_Type-if01-event-kbd -> ../event9
usb-Razer_Razer_Pro_Type-if02-event-kbd -> ../event13
usb-Razer_Razer_Pro_Type-mouse -> ../mouse1
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

dssouza-ti, 2020-11-23:
Hello. I also have a Razer Pro type. Don't know if it's important, but my product number is a little different:

- Name: Razer Pro Type
- Product Number: RZ03-03070200
- Model Number: RC30-030702

Mine is connected via Bluetooth, so how can I collect the device information and the PCAP needed to help provide support?

azhard, 2021-05-02:
I have a pro type as well, any info I can provide to help push this forward?

phlet, 2021-05-04:
Same here !
How can we help ?

Hedronmx, 2021-06-28:
Got a type too, what should we do :D

justintime4tea, 2022-09-11:
I have a Pro Type as well. I'll try to get a pcap tonight and try to help where I can. I'm also a software engineer and interested in the code to make this all work. Might be able to contribute more than just pcap... We'll see. I have several RAZER peripherals and my main rig is a Blade so I might be able to help quite a bit.

acerspyro, 2022-10-09:
I'm wondering if the Pro Type Ultra would also be supported by the same changes.

I just purchased one (hasn't arrived yet) and I need to be able to disable fn lock.

belafdil-chakib, 2022-10-19:
> I'm wondering if the Pro Type Ultra would also be supported by the same changes.
> 
> I just purchased one (hasn't arrived yet) and I need to be able to disable fn lock.

I just got my pro type ultra and it is not supported. If I have time I'll try to implement it and PR it

mneilly, 2023-04-11:
Attaching captures for power saving settings.

[razer-pro-type-keyboard-power-saving.zip](https://github.com/openrazer/openrazer/files/11197412/razer-pro-type-keyboard-power-saving.zip)

z3ntu, 2023-04-22:
Looks like a regular device with transaction_id=0x1f.

Supported: razer_chroma_misc_set_idle_time, razer_chroma_misc_get_charging_status, razer_chroma_misc_get_battery_level

I don't see any RGB in the pcap, does the device not have that?

dssouza-ti, 2023-04-22:
> I don't see any RGB in the pcap, does the device not have that?

Yes, it's only has white LEDs.

z3ntu, 2023-04-22:
But then I'd guess you can turn the LED on and off, and maybe adjust brightness also?

acerspyro, 2023-04-22:
You can still apply patterns and effects to these keyboards IIRC