openrazer/openrazer #1240: Support for Razer Cynosa V2

nicooo, 2020-08-29:
### Device Info

* Name: Razer Cynosa V2
* Product Number: RZ03-0340
* Model Number: RZ03-03400200-R3U1

### Device Information
```
Bus 001 Device 002: ID 1532:025e Razer USA, Ltd Razer Cynosa V2
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x025e 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Cynosa V2
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
```

```
total 0
drwxr-xr-x 2 root root 240 2020-08-29 12:02 ./
drwxr-xr-x 4 root root 680 2020-08-29 12:02 ../
lrwxrwxrwx 1 root root  10 2020-08-29 11:40 usb-Burr-Brown_from_TI_USB_Audio_DAC-event-if02 -> ../event11
lrwxrwxrwx 1 root root  10 2020-08-29 11:40 usb-Logitech_G_Pro_Gaming_Mouse_108B385C3437-event-if01 -> ../event12
lrwxrwxrwx 1 root root   9 2020-08-29 11:40 usb-Logitech_G_Pro_Gaming_Mouse_108B385C3437-event-mouse -> ../event9
lrwxrwxrwx 1 root root  10 2020-08-29 11:40 usb-Logitech_G_Pro_Gaming_Mouse_108B385C3437-if01-event-kbd -> ../event10
lrwxrwxrwx 1 root root   9 2020-08-29 11:40 usb-Logitech_G_Pro_Gaming_Mouse_108B385C3437-mouse -> ../mouse1
lrwxrwxrwx 1 root root   9 2020-08-29 11:40 usb-Razer_Razer_Cynosa_V2-event-if01 -> ../event6
lrwxrwxrwx 1 root root   9 2020-08-29 11:40 usb-Razer_Razer_Cynosa_V2-event-kbd -> ../event2
lrwxrwxrwx 1 root root   9 2020-08-29 11:40 usb-Razer_Razer_Cynosa_V2-if01-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 2020-08-29 11:40 usb-Razer_Razer_Cynosa_V2-if02-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 2020-08-29 11:40 usb-Razer_Razer_Cynosa_V2-if02-mouse -> ../mouse0
```

jcanso, 2020-09-11:
I have no idea what I'm doing, so here are two files for somebody to check and see if they're useful at all.  One is of the static lighting being switched on and off a couple of times and the other is of the spectrum cycling being switched on and allowed to run for about a minute.
[image]
[test_pcaps.zip](https://github.com/openrazer/openrazer/files/5211111/test_pcaps.zip)

If these are alright, I'll try out doing some more (do I have to do every effect for every colour?  Do I have to do anything with the customization tab?)

sorry for being such a noob

z3ntu, 2020-09-13:
Thanks, the pcaps are fine. This device uses [this for brightness](https://github.com/openrazer/openrazer/blob/master/driver/razerkbd_driver.c#L1621-L1622) and [this for the effects](https://github.com/openrazer/openrazer/blob/master/driver/razerkbd_driver.c#L1828-L1829) if someone wants to take a shot at implementing support for this device :)

jcanso, 2020-09-15:
well hot damn, after a bunch of trial and error and with the lines mentioned by @z3ntu, it looks like I have something that at least sort of works.  All of the effects that I tested so far have worked anyway!

jcanso, 2020-09-16:
#1257 after sorting out the daemon and the udev rules, then fighting with the code formatting, this is what I've got

commit 94d7428c (2020-10-11): Add support for Razer Cynosa V2

Fixes #1240
PR #1257 (2020-10-11): Add support for Razer Cynosa V2
Hopefully I managed to tick all the boxes.

Fixes #1240
closed 2020-10-11 by z3ntu