openrazer/openrazer #1241: Viper Ultimate Wireless not correctly detected in wireless mode

ddavini, 2020-08-29:
### Distribution

Xubuntu 20.04

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I've installed 2.8.0 and I tried with Polychromatic 0.3.12-stable and Razergenie 0.8.1 with the same result. My Viper Ultimate Wireless is correctly detected and the settings are read correctly only if connected with USB cable, while it is not with the dongle in wireless mode.

Wireless mode
[image]
[image]

Wired mode
[image]
[image]

nightsky30, 2020-08-29:
Please see #1164. Testing and an idea of what the appropriate timing values need to be set at would be great :)

ddavini, 2020-08-30:
No problem, I'll see what I can do. I have a Windows 7 VM somewhere, I don't know if razer stuff works still on 7 but I'll give it a try.

ddavini, 2020-08-30:
The `RAZER_NEW_MOUSE_RECEIVER` values seem to be working fine for me:

```
static int razer_get_report(struct usb_device *usb_dev, struct razer_report *request_report, struct razer_report *response_report)
{
    switch (usb_dev->descriptor.idProduct) {
    // These devices require longer waits to read their firmware, serial, and other setting values
    case USB_DEVICE_ID_RAZER_LANCEHEAD_WIRELESS_RECEIVER:
    case USB_DEVICE_ID_RAZER_LANCEHEAD_WIRELESS_WIRED:
    case USB_DEVICE_ID_RAZER_MAMBA_WIRELESS_RECEIVER:
    case USB_DEVICE_ID_RAZER_MAMBA_WIRELESS_WIRED:
        return razer_get_usb_response(usb_dev, 0x00, request_report, 0x00, response_report, RAZER_NEW_MOUSE_RECEIVER_WAIT_MIN_US, RAZER_NEW_MOUSE_RECEIVER_WAIT_MAX_US);
        break;
    case USB_DEVICE_ID_RAZER_VIPER_ULTIMATE_WIRED:
    case USB_DEVICE_ID_RAZER_VIPER_ULTIMATE_WIRELESS:
        return razer_get_usb_response(usb_dev, 0x00, request_report, 0x00, response_report, RAZER_NEW_MOUSE_RECEIVER_WAIT_MIN_US, RAZER_NEW_MOUSE_RECEIVER_WAIT_MAX_US);
        break;    
    default:
        return razer_get_usb_response(usb_dev, 0x00, request_report, 0x00, response_report, RAZER_MOUSE_WAIT_MIN_US, RAZER_MOUSE_WAIT_MAX_US);
    }
}
```

It's working well enough for me, but I can set up a Windows VM and sniff the response times if you think it's worth it.

Thanks for your help!

ddavini, 2020-09-06:
Hi, I wanted to add that with `RAZER_NEW_MOUSE_RECEIVER`, sometimes right after the system starts **Polychromatic Tray Applet** shows the wrong DPI value, something like 54000 (LOL :)). If you refresh with **Refresh Device List** it gets it right all the times, so far anyway.  So now I decided to try:
- 33000 Max
- 31000 Min

Was still happening with 32000 Max.

Also I don't think the such high values are necessary when the device is in wired mode. But I just did a quick test, I don't use my mouse that way.

I wonder if it happens with other wireless mice too, anyway. I'll test it for a while and let you know.

ddavini, 2020-09-08:
Okay, I got bored of tip toeing into it and I cranked it up to 60000. Now it happens very rarely but it still happens. I mean, it happened once after I repeatedly restarted the daemon.  

I also noticed that in Polychromatic the battery is always 100% even though it says otherwise when you restart the daemon from the tray applet. 

Would it be possible to have the values configurable in a config file?

ddavini, 2020-09-08:
One more thing then I'll disappear.

On my system `RAZER_VIPER_MOUSE_RECEIVER_WAIT_MIN_US 30000` and `RAZER_VIPER_MOUSE_RECEIVER_WAIT_MAX_US 60000` makes it definitely more reliable with no evident side effects.

PR #1164 (2020-09-26): Viper/Ultimate Cleanup
The Viper and Viper Ultimate currently only generate device files and reports for breath, static, and none effects in the driver.

Viper / Viper Ultimate:
 - Cleaned up mouse.py by removing method references to wave effect
 - Removed WAVE_DIRS, HAS_MATRIX, and MATRIX_DIMS as wave wasn't being used (not supported), and they only appear to have Logo LEDs
 - Updated to add support for spectrum and reactive effects
 - Cleaned up fake drivers to reflect the current effect functionality in the driver

Viper Ultimate:
- Added missing battery methods to match driver
- Added miss [... cut]
closed 2020-09-08 by ddavini