openrazer/openrazer #1242: Unable to launch package once installed -- "openrazer

carlcrott, 2020-08-31:
### Distribution

Ubuntu 18.04

### Support

- [X] Have you tried rebooting?
- [X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Followed the directions here:
https://openrazer.github.io/#ubuntu

but when I attempt to launch the tool with the installed package, none is found.

$ openrazer
$ openrazer-meta

Seems weird that the installed package name doesnt launch the installed package...

lah7, 2020-08-31:
The only component of OpenRazer that can be launched is the daemon, which is `openrazer-daemon`. The API is available via Python/D-Bus. For a user interface, [install an application such as Polychromatic or RazerGenie](https://github.com/openrazer/openrazer#applications). The daemon normally automatically starts when an application / API is used.

This command will be useful at listing at what's installed:

    dpkg-query -L <package_name>

[`openrazer-meta` is a metapackage](https://www.debian.org/blends/hamradio/get/metapackages) which installs **`openrazer-daemon`**, `python3-openrazer`, `openrazer-doc` and `openrazer-driver-dkms`.

carlcrott, 2020-09-01:
Following directions here:

https://software.opensuse.org//download.html?project=hardware%3Arazer&package=razergenie

for ubuntu 18.04:

```
echo 'deb http://download.opensuse.org/repositories/hardware:/razer/xUbuntu_18.04/ /' | sudo tee /etc/apt/sources.list.d/hardware:razer.list
curl -fsSL https://download.opensuse.org/repositories/hardware:razer/xUbuntu_18.04/Release.key | gpg --dearmor | sudo tee /etc/apt/trusted.gpg.d/hardware:razer.gpg > /dev/null
sudo apt update
sudo apt install razergenie
```

Fixed all my problems!

Thanks @lah7

closed 2020-09-01 by carlcrott