openrazer/openrazer #1243: Making a Slackware package question

mdinslage, 2020-08-31:
Hi all, now that 2.8.0 supports Basilisk mouse, I went ahead and made a Slackware package for openrazer, I plan on submitting it to www.slackbuilds.org when version 15 goes stable.  I have one question though...

During the build process, once the kernel modules are built, I move them from /usr/src/openrazer to /lib/modules/VERSION/kernel/drivers/hid which is the correct location for Slackware.   After that can I safely delete the /usr/src/openrazer directory or are there other files in there that are needed?

z3ntu, 2020-09-01:
Hi, that depends on whether you want the package to contain the built kernel modules or if you'll ship the sources which will be built with dkms. In the former case you just need the *.ko files and in the latter just the sources in /usr/src . Afaik just Solus has binary packages for the kernel modules, the rest of the distros is handled via dkms because I can't guarantee a timely rebuild for every kernel update that's coming down in the distro. So basically it depends what your distro is doing in other packages.

mdinslage, 2020-09-01:
Slackware will just need the modules then, its standard practice to build and include modules in your SlackBuild.  Yeah the package will need to be rebuilt every time there is a kernel update, but that is not very often in Stable Slackware, only if there's a vulnerability.   Plus the Slackware community will already know this.  Many Slackware packages are like this,  the NVIDIA driver package comes to mind.   

Thanks for this project,  It's nice having my mouse a static color now  :)

closed 2020-09-01 by mdinslage