openrazer/openrazer #1245: Support for [device name here]

v21cesc, 2020-09-05:
### Device Info

* Name: Razer Viper Mini 8500dpi
* Product Number: RZ01-03250300
* Model Number: RZ01-0325 GAMING MOUSE

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
Bus 001 Device 008: ID 1532:008a Razer USA, Ltd Razer Viper Mini
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x008a 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Viper Mini
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)

```
Please post the output of
    ls -lah /dev/input/by-id/
```

usb-MSI_MYSTIC_LIGHT_A02020043004-event-if00 -> ../event25
usb-Razer_Razer_Viper_Mini-event-if01 -> ../event6
usb-Razer_Razer_Viper_Mini-event-mouse -> ../event4
usb-Razer_Razer_Viper_Mini-if01-event-kbd -> ../event5
usb-Razer_Razer_Viper_Mini-if02-event-kbd -> ../event9
usb-Razer_Razer_Viper_Mini-mouse -> ../mouse0

### Packet Captures
<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

Coming soon

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

Coming soon

nightsky30, 2020-09-05:
See #1163

v21cesc, 2020-09-05:
> See #1163

Thanks. I've tried the latest commit and it's working perfectly with this mouse.

roadhoghook, 2020-09-06:
> > See #1163
> 
> Thanks. I've tried the latest commit and it's working perfectly with this mouse.

do you mind tell me how to edit the PKGBUILD file, I've been trying for a long time and failing T_T

z3ntu, 2020-09-06:
Closing as duplicate of #1062

closed 2020-09-06 by z3ntu

roadhoghook, 2020-09-06:
> Closing as duplicate of #1062

do you mind tell me how to edit the PKGBUILD file, I've been trying for a long time and failing T_T

v21cesc, 2020-09-06:
> > Closing as duplicate of #1062
> 
> do you mind tell me how to edit the PKGBUILD file, I've been trying for a long time and failing T_T

I'm not sure. Maybe, however, you can try this one: https://github.com/openrazer/openrazer/wiki/Building-a-package