openrazer/openrazer #1247: Support for Razer Naga Pro

teeehhias, 2020-09-09:
### Device Info

* Name: Razer Naga Pro
* Product Number: RZ01-0342
* Model Number: RZ01-03420100

### Device Information

`Bus 001 Device 011: ID 1532:0090 Razer USA, Ltd Razer Naga Pro
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0090 
  bcdDevice            1.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0086
    bNumInterfaces          5
    bConfigurationValue     1
    iConfiguration          4 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     179
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        4
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     325
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     348
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1`

`lrwxrwxrwx 1 root root   10 Sep  9 12:36 usb-RAZER_Razer_Mouse_Dock-event-if01 -> ../event30
lrwxrwxrwx 1 root root   10 Sep  9 12:36 usb-RAZER_Razer_Mouse_Dock-event-mouse -> ../event26
lrwxrwxrwx 1 root root   10 Sep  9 12:36 usb-RAZER_Razer_Mouse_Dock-if01-event-kbd -> ../event27
lrwxrwxrwx 1 root root   10 Sep  9 12:36 usb-RAZER_Razer_Mouse_Dock-if02-event-kbd -> ../event31
lrwxrwxrwx 1 root root    9 Sep  9 12:36 usb-RAZER_Razer_Mouse_Dock-mouse -> ../mouse0
lrwxrwxrwx 1 root root   11 Sep  9 12:36 usb-Razer_Razer_Naga_Pro_000000000000-event-if01 -> ../event263
lrwxrwxrwx 1 root root   11 Sep  9 12:36 usb-Razer_Razer_Naga_Pro_000000000000-event-if03 -> ../event268
lrwxrwxrwx 1 root root   11 Sep  9 12:36 usb-Razer_Razer_Naga_Pro_000000000000-event-mouse -> ../event257
lrwxrwxrwx 1 root root   11 Sep  9 12:36 usb-Razer_Razer_Naga_Pro_000000000000-if01-event-kbd -> ../event258
lrwxrwxrwx 1 root root   11 Sep  9 12:36 usb-Razer_Razer_Naga_Pro_000000000000-if02-event-kbd -> ../event266
lrwxrwxrwx 1 root root    9 Sep  9 12:36 usb-Razer_Razer_Naga_Pro_000000000000-mouse -> ../mouse1`

binarypie, 2020-09-17:
I can help with the pcap what else do you need beyond this? 
https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

cottagepainter, 2020-09-18:
I just got a Razer Naga Pro. Running Arch-Linux. Anything i can do to help?

YoonAddicting, 2020-09-19:
I also got the mouse running Ubuntu 20.04, and can lend a hand if any needed.

teeehhias, 2020-09-21:
> I can help with the pcap what else do you need beyond this?
> https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

This would be great, I tried to, but somehow when I pass the mouse to one of my VM's I can not perform any actions with it. Couldn't figure it out by now.

Also, the mouse dock (bought separately) seems not to be recognized correctly. I've read this is mouse specific?

Here is the lsusb -d 1532: -v output if that's relevant:

- Product Number: RC30-03050200-0000
- Model Number: RC30-030502

```Bus 001 Device 011: ID 1532:007e Razer USA, Ltd RC30-030502 Mouse Dock
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x007e RC30-030502 Mouse Dock
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
```

YoonAddicting, 2021-01-04:
I noticed that the Mouse Dock is also shared with the Basilisk Ultimate and DeathAdder V2 Pro as well as the Naga Pro. Seems like the functionality for the dock is almost implemented in PR https://github.com/openrazer/openrazer/pull/1321

Mushoz, 2021-01-20:
@binarypie the pcap files and synapse screenshots are required to get a device supported. To get a pcap file that is useful, follow the instructions that are given here: https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol
Make sure you save the data as a .pcapng file as requested in said documentation.

razc411, 2021-01-25:
I'm also interested in getting this mouse working, so I've followed the directions and this is the pcap produced + pictures of synapse. Tell me if there's anything wrong with it. I hit every mouse action during this pcap.

https://drive.google.com/file/d/1-7v6Aqq2-kcj-DYtnYUwq32m0FRBUsyy/view?usp=sharing (pcap)
https://drive.google.com/file/d/19lyY6GCWroWAAbd3Dx7TDKdq-vmWQuNt/view?usp=sharing (pictures)

seathasky, 2021-01-25:
> I'm also interested in getting this mouse working, so I've followed the directions and this is the pcap produced + pictures of synapse. Tell me if there's anything wrong with it. I hit every mouse [...]

Godlike!! Support soon I hope " crosses fingers"

cottagepainter, 2021-01-28:
@razc411 Did you hit the button on the bottom of the mouse (which switches your profile)? Also don't forget that the mouse wheel has 'tilt left' and 'tilt right'. I didn't check the pcap, just asking.

qauff, 2021-01-30:
> I'm also interested in getting this mouse working, so I've followed the directions and this is the pcap produced + pictures of synapse. Tell me if there's anything wrong with it. I hit every mouse [...]

I’m curious, do these pcaps include Bluetooth connection?

razc411, 2021-01-30:
@cottagepainter @qauff I did not hit the switch button or bluetooth and was using 2.4 ghz. If it is needed, I can provide a bluetooth pcap and one that includes the switch toggle.

qauff, 2021-01-30:
> @cottagepainter @qauff I did not hit the switch button or bluetooth and was using 2.4 ghz. If it is needed, I can provide a bluetooth pcap and one that includes the switch toggle.

If you could! Please! That’d be awesome of you!🤓

qauff, 2021-02-01:
The productID for the Naga Pro via Bluetooth is `0x0092` from what I understand though, I don't think openrazer supports any Bluetooth connections thus far (I could be wrong), so supporting bluetooth could be a bigger endeavour than just simply adding support for a new USB device.

ProjectInfinity, 2021-02-19:
Is there any update on this? What's the stopper for adding support for the Naga Pro?

qauff, 2021-02-19:
@ProjectInfinity developers with the mouse :)

GeoDoX, 2021-02-19:
Don’t mean to sound rude but isn’t that the point of the pcap, so devs don’t need the hardware? Just as easy for them to get the pcap if they had the hardware isn’t it?

If the pcap has been supplied can we remove the needs pcap tag?

seathasky, 2021-02-20:
> Don’t mean to sound rude but isn’t that the point of the pcap, so devs don’t need the hardware? Just as easy for them to get the pcap if they had the hardware isn’t it?
> 
> If the pcap has been supplied can we remove the needs pcap tag?

Unfortunately it seems as the devs of this project are inactive. I think their last release was 4 months ago so I wouldnt expect much from them. This could have been completed in a weekend but dont think there is anyone working on it. Just a bunch of us talking about it and no actual work being done haha. Same goes with razer-macos community. Not much progress being made from the actual devs and only from people outside the project trying to get things to work. https://github.com/1kc/razer-macos/issues/189

cottagepainter, 2021-02-21:
@seathasky the latest commit to master was today, and theirs been 28 commits in the past 4 months, many of them for device support...so the devs are still active! Be patient.

In the meantime, customize your mouse on a window PC and it works just fine for now.

z3ntu, 2021-02-27:
Thanks @razc411 

If someone wants to look into implementing this, the device supports DPI stages, custom frame uses `razer_chroma_extended_matrix_set_custom_frame` with `MATRIX_DIMS = [1, 3]` . Transaction id is `0x1f` for everything. For the rest of the effects you'll have to do trial & error and see what works, Synapse 3 doesn't use those commands anymore so this info isn't in the pcap.
If you have any specific questions feel free to ask.

ProjectInfinity, 2021-04-02:
I'm not sure if it's allowed but I'll gladly put up a bounty for adding support for the naga pro. It's an excellent mouse but without ability to customise it without using a Windows computer, it's a bit crippled.

BrandonArp, 2021-04-02:
I'm also in for adding to a bounty.

CarlOlson, 2021-04-03:
+1 for a bounty

mikhailswift, 2021-04-04:
I just received this mouse yesterday.  I don't have experience developing on OpenRazer but I'm gonna start working on implementing this in OpenRazer.

juupajuupa, 2021-05-09:
Hey! Any updates on this topic?

mikhailswift, 2021-05-10:
I've unfortunately not had much time to spend on working on this, but I am still tracking it on my to-do list.  Will hopefully be able to sink some time into it this week/weekend.

julmajustus, 2021-07-24:
Is there any progress on this? Any tricks to get the mouse work correctly?

seathasky, 2021-07-24:
No. No one has proceeded in finishing this (yet)

GeoDoX, 2021-09-09:
Bump, anyone working on this yet? @mikhailswift ?

j-maynard, 2021-09-22:
I've been looking at the BlackWidow Pro V3 and The Naga Pro...  I should probably draw peoples attention to the fact that if you have your Hyperspeed Dongle used for Both Mouse and Keyboard the the mouse in my case the Naga Pro is a reported to XInput as Razer BlackWidow V3 Pro Mouse... See the screenshot:

[image]

My guess is that if you do the same with the Razer HyperSpeed Dongle which comes with the Naga the device will report as a Razer Naga Pro Keyboard.  I've not tested that yet.  I've cross posted this part of the issue with the Support for the BlackWidow Pro V3 issue.

LEOMODE, 2021-09-24:
Hoping to get Naga Pro working on Mac OS soon.

seathasky, 2021-09-24:
People have been hope since release. No one is officially working on it
unfortunately

GeoDoX, 2021-09-24:
@LEOMODE I believe the razer-macos community and the open-razer community work pretty close together. When a device gets support on one, the other usually picks it up. Hopefully one of the communities picks up support for this mouse soon.

joniler, 2021-10-11:
BUMP! Any progress here? I would help, but I'm just a humble web developer... =\

joniler, 2021-10-15:
I will try my hand at working on it if someone c an send me a pcap

GeoDoX, 2021-10-17:
@razc411 the best way to learn is to look at PRs of similar devices. See what files are added/modified and what code is in each.

As for testing, you just need to build the debs, install, and restart. For example on Ubuntu;
`./scripts/build_debs.sh
sudo apt install ./dist/*.deb`

For other examples; https://github.com/openrazer/openrazer/wiki/Building-a-package
Just note, you don’t need to clone the repo as you have already done so to make the modifications.

voshond, 2021-12-20:
bump, would love to see support for this 👍

Ersmipe, 2021-12-20:
We all would and it seems that @AliceGrey managed to make naga work. We should probably try her branch.

seathasky, 2021-12-20:
People have been asking for this since launch and no one will do it. I wouldn't get your hopes up.

GeoDoX, 2021-12-21:
> People have been asking for this since launch and no one will do it. I wouldn't get your hopes up.

#1692

Nezmin2, 2022-02-18:
Are there any updates on this issue? I just purchased the Naga Pro and it's not recognized.

hackeraf, 2022-02-23:
Trying to use @AliceGrey's branch but I'm not sure the proper way to install this. Please help.

sudoimposed, 2022-02-23:
I've just been hoping it would be merged into the main package, but it's
been a couple months and nothing yet.  I'm not really that familiar with
standard practices, is it something that usually happens eventually or am I
better off trying to use the branch?

hackeraf, 2022-02-23:
I'm using @AliceGrey's branch and I still don't have the Naga Pro recognized:

[image]

ciferkey, 2022-03-16:
@hackeraf I installed the branch to test and it detected the mouse successfully
[image]

LEOMODE, 2022-03-17:
> @hackeraf I installed the branch to test and it detected the mouse successfully [image]
> 
> Note it worked wired and with 2.4 but not BT. Were you trying with BT?

Nice, did it work with the USB dongle also? I would like to test it out if it becomes available in Mac OS.

ciferkey, 2022-03-17:
Yes the 2.4ghz dongle detected! I did not test key remapping. It looks like Polychromatic is only for color. What app can be used to test key rebinding?

marcus13345, 2022-04-04:
@hackeraf Im also trying to install this branch locally, and the linked instructions arent making a lot of sense to me. Im on fedora, have the repo cloned / checked out to the right branch. I downloaded the spec file, but im not sure where to go from there. the guide says to change the `define gitcommit` line and the `source0` line, but they currently reference tar.gz / tar.xz files online. Seemingly id want to link it to my local version, but the guide does not show an example for that.

PR #1692 (2022-04-07): Add Razer Naga Pro Wired/Wireless Support 
Add Razer Naga Pro Support

Also Fixes #1247 (once completed)

So far what's working:

    DPI
    Polling
    Scroll Wheel LED
    Logo LED
    Pretending the left thumbgrid is the backlight LED
    Battery monitoring
    Wireless

What's not working:

    Addressing just the left thumbgrid LED independently
    Spectrum, Reactive, Breath for left thumbgrid
    Tilt wheel?
closed 2022-04-07 by z3ntu