openrazer/openrazer #1250: Razer BlackWidow 2019 not working on arch

yistarostin, 2020-09-11:
### Device Info

* Name: <Razer BlackWindow 2019>
* Product Number: <Razer BlackWindow 2019>
* Model Number: <Razer BlackWindow 2019>

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```

Bus 003 Device 003: ID 1532:0241 Razer USA, Ltd BlackWidow (2019)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0241 BlackWidow (2019)
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     179
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0016  1x 22 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

```
Please post the output of
    ls -lah /dev/input/by-id/
```
total 0
drwxr-xr-x 2 root root 180 Sep 11 16:44 .
drwxr-xr-x 4 root root 620 Sep 11 16:44 ..
lrwxrwxrwx 1 root root  10 Sep 11 16:44 usb-Kingston_HyperX_Pulsefire_FPS_Gaming_Mouse-event-mouse -> ../event23
lrwxrwxrwx 1 root root   9 Sep 11 16:44 usb-Kingston_HyperX_Pulsefire_FPS_Gaming_Mouse-mouse -> ../mouse1
lrwxrwxrwx 1 root root  10 Sep 11 16:44 usb-Razer_Razer_BlackWidow-event-if01 -> ../event19
lrwxrwxrwx 1 root root  10 Sep 11 16:44 usb-Razer_Razer_BlackWidow-event-kbd -> ../event17
lrwxrwxrwx 1 root root  10 Sep 11 16:44 usb-Razer_Razer_BlackWidow-if01-event-kbd -> ../event18
lrwxrwxrwx 1 root root  10 Sep 11 16:44 usb-Razer_Razer_BlackWidow-if02-event-mouse -> ../event22
lrwxrwxrwx 1 root root   9 Sep 11 16:44 usb-Razer_Razer_BlackWidow-if02-mouse -> ../mouse0

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

z3ntu, 2020-09-13:
Can you please provide more details about your problem?

yistarostin, 2020-09-13:
So my BlackWindow 2019 is plugged in, but in the polychromatic controller says: "Device not recocnized". Here is /openrazer/logs/razer.log file contect:
 
2020-09-12 16:03:59 | razer                          | CRITICAL | User is not a member of the plugdev group
2020-09-12 16:03:59 | razer                          | CRITICAL | Please run the command 'sudo gpasswd -a $USER plugdev' and then reboot!
2020-09-12 16:04:17 | razer                          | CRITICAL | User is not a member of the plugdev group
2020-09-12 16:04:17 | razer                          | CRITICAL | Please run the command 'sudo gpasswd -a $USER plugdev' and then reboot!
2020-09-12 16:05:22 | razer                          | CRITICAL | User is not a member of the plugdev group
2020-09-12 16:05:22 | razer                          | CRITICAL | Please run the command 'sudo gpasswd -a $USER plugdev' and then reboot!
2020-09-12 16:05:51 | razer                          | INFO     | Initialising Daemon (v2.8.0). Pid: 1304
2020-09-12 16:05:51 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2020-09-12 16:05:51 | razer                          | INFO     | Serving DBus
2020-09-12 19:09:00 | razer                          | INFO     | Initialising Daemon (v2.8.0). Pid: 1728
2020-09-12 19:09:00 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2020-09-12 19:09:00 | razer                          | INFO     | Serving DBus
2020-09-12 19:11:12 | razer                          | INFO     | Initialising Daemon (v2.8.0). Pid: 1412
2020-09-12 19:11:12 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2020-09-12 19:11:12 | razer                          | INFO     | Serving DBus
2020-09-12 20:00:42 | razer                          | INFO     | Initialising Daemon (v2.8.0). Pid: 1399
2020-09-12 20:00:42 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2020-09-12 20:00:42 | razer                          | INFO     | Serving DBus
2020-09-13 10:09:31 | razer                          | INFO     | Initialising Daemon (v2.8.0). Pid: 1450
2020-09-13 10:09:31 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2020-09-13 10:09:31 | razer                          | INFO     | Serving DBus

Keyboard itself(typing) is working fine, keyboad PID is recognized (1532:0241).

z3ntu, 2020-09-13:
Can you please run `sudo dkms install openrazer-driver/2.8.0` ?

Samega7Cattac, 2020-09-24:
@yistarostin 
had the same issue
I think it was bc I had some packages `-git` and other stable
reinstalled everything with `-git` and the problem was solved

yistarostin, 2020-09-30:
I am sorry for ignoring for a long periof of time. I had tried this keyboard on other arch machine, faced the same issue
@z3ntu  thanks for the reply, done that, but it didn't fix the issue. 
``dkms status`` output:
> openrazer-driver, 2.8.0: added

@Samega7Cattac Thanks for advice, but i didn't help (I had polychromatics and openrazer without ``-git``, i have reinstalled them with ``-git``

z3ntu, 2020-10-03:
The dkms status output should look like this:
```
openrazer-driver, 2.8.0, 5.8.13-arch1-1, x86_64: installed
```
Is there any output from the `dkms install` command I posted before?

yistarostin, 2021-05-03:
I am really sorry for opening such a stupid issue, the problem was in kernel version (i had wrong dkms headers installed, so kernel patch didn't work). Now it works as expected! Thank you, very useful piece of software

closed 2021-05-03 by yistarostin