openrazer/openrazer #1251: Support for [Razer Kraken Essential Gaming Headset]!

tgwow, 2020-09-11:
### Device Info

* Name: <Razer Kraken Essential Gaming Headset>
* Product Number: <RZ04-01720100>
* Model Number: <RZ04-0172>

### Device Information
```
Bus 001 Device 002: ID 1532:005e Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x005e 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
```

```
lrwxrwxrwx 1 root root  10 set 11 14:24 usb-BY_Tech_Usb_Gaming_Keyboard-event-if01 -> ../event12
lrwxrwxrwx 1 root root   9 set 11 14:24 usb-BY_Tech_Usb_Gaming_Keyboard-event-kbd -> ../event9
lrwxrwxrwx 1 root root  10 set 11 14:24 usb-BY_Tech_Usb_Gaming_Keyboard-if01-event-mouse -> ../event13
lrwxrwxrwx 1 root root   9 set 11 14:24 usb-BY_Tech_Usb_Gaming_Keyboard-if01-mouse -> ../mouse3
lrwxrwxrwx 1 root root  10 set 11 14:24 usb-Chicony_Electronics_Co._Ltd._HD_WebCam_0001-event-if00 -> ../event20
lrwxrwxrwx 1 root root  10 set 11 14:24 usb-Razer_Razer_Abyssus_2000-event-mouse -> ../event15
lrwxrwxrwx 1 root root   9 set 11 14:24 usb-Razer_Razer_Abyssus_2000-mouse -> ../mouse0
```

z3ntu, 2020-09-13:
It doesn't look like this device has a USB connection or any lighting effects?

tgwow, 2020-09-20:
> It doesn't look like this device has a USB connection or any lighting effects?

There's no USB connection, it's a p3 plug and doesn't have lighting effects

z3ntu, 2020-09-20:
Then there's nothing we can support in OpenRazer with this device. Closing.

closed 2020-09-20 by z3ntu