openrazer/openrazer #1252: Razer Naga Trinity: Unable to draw matrix

lah7, 2020-09-14:
Multiple users in https://github.com/polychromatic/polychromatic/issues/266 have reported this exception being thrown when attempting to draw a matrix using the Python library for the **Razer Naga Trinity**.

```
ERROR:dbus.connection:Unable to set arguments (b'\x00\x00\x02\x00\xff\x00\x00\x00\x00\x00\x00\x00',) according to signature None: <class 'ValueError'>: embedded null byte
Traceback (most recent call last):
  File "step-through-matrix.py", line 40, in <module>
    device.fx.advanced.draw()
  File "/usr/lib/python3.8/site-packages/openrazer/client/fx.py", line 600, in draw
    self._draw(bytes(self.matrix))
  File "/usr/lib/python3.8/site-packages/openrazer/client/fx.py", line 592, in _draw
    self._lighting_dbus.setKeyRow(ba)
  File "/usr/lib/python3.8/site-packages/dbus/proxies.py", line 141, in __call__
    return self._connection.call_blocking(self._named_service,
  File "/usr/lib/python3.8/site-packages/dbus/connection.py", line 643, in call_blocking
    message.append(signature=signature, *args)
ValueError: embedded null byte
```

To reproduce:

```
import openrazer.client as rclient
a = rclient.DeviceManager()
device = devman.devices[0]     # Assume first device
device.fx.advanced.matrix[0, 0] = [0, 255, 0]
device.fx.advanced.draw()
```

z3ntu, 2020-09-14:
Normally when a D-Bus method can't be found this exception gets thrown:
```
dbus.exceptions.UnknownMethodException: org.freedesktop.DBus.Error.UnknownMethod: Unknown method: hasMatrix is not a valid method of interface razer.device.misc
```
But for some reason it seems that when the we're trying to send binary data to a method but that method doesn't exist we get the exception
```
ValueError: embedded null byte
```

So actually the set_key_row method is missing from dbus - either driver has missing values or daemon has invalid HAS_MATRIX = True - the same seems to apply to other devices as well :(

z3ntu, 2020-09-14:
A.k.a a bunch of devices that declare `HAS_MATRIX = True` don't actually have the `set_key_row` method and fail because of that. Technically this is also the same as #683 but the original issue there was that the daemon had a typo causing the `set_key_row` method not to be added to dbus

lah7, 2020-10-22:
So, based on the fake driver, these devices are affected. These specify the capability `lighting_led_matrix`.

* [ ] Razer BlackWidow Essential
  * `dbus.exceptions.UnknownMethodException: org.freedesktop.DBus.Error.UnknownMethod: Unknown method: setCustom is not a valid method of interface razer.device.lighting.chroma`
* [ ] Razer Viper Ultimate (Wired)
  * `embedded null byte`
* [ ] Razer Blade Stealth (2019)
  * `embedded null byte`
* [ ] Razer Viper Ultimate (Wireless)
  * `embedded null byte`
* [ ] Razer Firefly Hyperflux 
  * `dbus.exceptions.UnknownMethodException: org.freedesktop.DBus.Error.UnknownMethod: Unknown method: setCustom is not a valid method of interface razer.device.lighting.chroma`
* [ ] Razer Viper
  * `embedded null byte`
* [ ] Razer Blade 15 (2018) Base Model
  * `embedded null byte`
* [ ] Razer BlackWidow Lite
  * `dbus.exceptions.UnknownMethodException: org.freedesktop.DBus.Error.UnknownMethod: Unknown method: setCustom is not a valid method of interface razer.device.lighting.chroma`
* [ ] Razer Naga Trinity
  * `embedded null byte`

<details>
<summary>The test script</summary>
<pre>
import openrazer.client
a = openrazer.client.DeviceManager()
for d in a.devices:
    if d.has("lighting_led_matrix"):
        d.fx.advanced.matrix[0,0] = [0,255,0]
        try:
            d.fx.advanced.draw()
        except Exception as e:
            print(d.name + ": " + str(e))
</pre>
</details>

z3ntu, 2020-11-01:
@nightsky30 I'm pretty sure Razer Viper should support `razer_chroma_extended_matrix_set_custom_frame` with transaction id 0x1f and matrix dimensions [1, 1] (so index 0, 0 only). I'm guessing Viper Ultimate should be the same but I don't have packet captures for that device.

closed 2020-11-01 by z3ntu

nightsky30, 2020-11-02:
@z3ntu Don't have the Viper or Viper Ultimate, but do have the Mini. Here's a sample packet capture:
[viper-mini_static_spectrum_static_spectrum_static.pcapng.zip](https://github.com/openrazer/openrazer/files/5472089/viper-mini_static_spectrum_static_spectrum_static.pcapng.zip)

Does appear to be using transaction ID 0x1f.

z3ntu, 2020-11-03:
@nightsky30 Okay could you please prepare a PR for these devices and add the neccessary methods to the driver & daemon?

PR #1305 (2020-11-01): Correct misconfigured matrix capabilities in the daemon
Fixes #1252.

* Devices that were throwing `embedded null byte` didn't have a `matrix_effect_custom` sysfs file, and so had `HAS_MATRIX` set to `False`.
* Devices that had this file had `set_key_row` and/or `set_custom_effect` added.

:heavy_check_mark:  The test script in that issue (against fake devices) passes without exceptions being thrown.
commit 5ac3d5f5 (2020-11-01): daemon: fix misconfigured matrix capabilities

Devices were throwing an exception that was:
- "embedded null byte"
- or "setCustom is not a valid method of interface"

This was due to devices missing a sysfs file in the driver. Or rather,
they had one (matrix_effect_custom), but was incorrectly specified here.

For devices that are missing a sysfs file, it's possible custom frames
are unsupported or was simply left out in the driver when the device was
added. Users of these devices should try amending their device in the
driver, as we don't have the hardware to check.

Fixes #1252
Addresses po [... cut]