openrazer/openrazer #1254: Support for Razer BlackShark V2 Headset

cyferc, 2020-09-15:
I'd like to enable and adjust sidetone on this headset. It does not have onboard persistent memory so sidetone is off by default and can only be enabled on Synapse 3.

### Device Info

* Name: Razer BlackShark V2
* Product Number: RZ04-03230100-R3U1
* Model Number: RZ04-03230100-R3U1

### Device Information
```
~$ lsusb -d 1532: -v
Bus 004 Device 002: ID 1532:0529 Razer USA, Ltd Razer USB Sound Card
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0529 
  bcdDevice            0.0e
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0136
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0072
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID            10
        wTerminalType      0x0201 Microphone
        bAssocTerminal          0
        bNrChannels             2
        wChannelConfig     0x0003
          Left Front (L)
          Right Front (R)
        iChannelNames           0 
        iTerminal               6 
[... 291 lines cut ...]
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0025  1x 37 bytes
        bInterval               1
```

```
~$ ls -lah /dev/input/by-id/
lrwxrwxrwx 1 root root   9 Sep 16  2020 usb-Razer_Razer_USB_Sound_Card_00000000-event-if03 -> ../event8
```

### Packet Captures
Will update soon.
<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
Will update soon.
<Screenshots of all available options as seen on Razer Synapse>

cyferc, 2020-09-18:
I uploaded screenshots and wireshark captures. Please let me know if you need anything else, I tried to capture all use cases.

z3ntu, 2020-09-20:
As far as I can tell all the audio effects you get with Synapse on Windows are purely software based so there's nothing we can send the hardware to enable these things. Additionally from what I put together the usb device 1532:0529 is Razer's generic "Razer USB Sound Card" which seems to only do USB to 3.5mm and nothing more? So there's nothing we can do with this device in OpenRazer afaik.

z3ntu, 2020-09-20:
Actually I've just googled what "sidetone" is and found this:

> Sidetone is sound picked up by your headset microphone that is then played back in real-time into the headset's speaker(s), acting as controlled feedback.

So basically a loopback from your mic into your headphones? This should be doable with just pulseaudio ("pulseaudio mic loopback" in e.g. Google)

cyferc, 2020-09-20:
Normal mic loopback is not the same as it has some lag, sidetone is instant without lag and is a hardware feature inside the USB sound card. All the wireshark captures I got are hardware features so should be able to be implemented in openrazer? I've creates a branch and will try implement this but still a noob when it comes to linux drivers.

This headset will no doubt be super popular as it got really good reviews so would be nice if openrazer supported it :)

cyferc, 2020-09-20:
Can you reopen this issue please? I'll try implement this in the next few weeks.

HarshitJoshi, 2020-12-16:
>  Additionally from what I put together the usb device 1532:0529 is Razer's generic "Razer USB Sound Card" which seems to only do USB to 3.5mm and nothing more? So there's nothing we can do with this device in OpenRazer afaik.

Please correct me if I'm wrong, but does this mean it cannot be supported at all with OpenRazer? I pulled down @cyferc's branch and was able to get it partially working in OpenRazor (just device recognition, no controls, only audio, no mic support). I'm new to this, but willing to spend some time over the weekend. Maybe @cyferc has some updates?

Thanks for this project guys!

ceiphr, 2021-01-21:
@HarshitJoshi Glad to hear you got it working. Could you elaborate on what pulseaudio settings you changed?

HarshitJoshi, 2021-01-21:
@ceiphr I have the Blackshark V2, to start off, you would want to pull down cyerc's repo (https://github.com/cyferc/openrazer) and build. Regarding the PulseAudio setup, the "Razer USB Sound Card" (at least for me) with Analog Stereo Duplex under the Configuration tab in Pulseaudio worked for me. Let me know if you need any further help. Bear in mind, all I've been able to do is minimal audio control and mic usage. No fancy sound/mixer controls.

ceiphr, 2021-01-26:
@Nyno-Kun IIRC on Ubuntu 20.10 I didn't seem to have any issues. Maybe @HarshitJoshi or @cyferc can help?

HarshitJoshi, 2021-01-28:
What errors did you get @Nyno-Kun. If you post them here, I'll take a look over the weekend.

Exerionius, 2024-09-27:
Any news on this or is it still "not planned"?

z3ntu, 2024-09-29:
@Exerionius You can also take a look at https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md how you could add support for this device yourself. Somebody with the device will have to do this.

Exerionius, 2024-09-30:
@z3ntu I was just planning on buying this model, but it doesn't make a lot of sense to spend money on something that is not properly supported by my system.

z3ntu, 2024-09-30:
If you want to support a company that cares about Linux, then it's probably best not to buy products from Razer anyways.